mne-tools/mne-python #3542: drop_channels silently passes when no channels are dropped

jona-sassenhagen, 2016-08-29:
Maybe it's only me who's this thick skulled, but I do this all the time:

`raw.drop_channels('EEG 000')`

And then wonder why my channel is still in there.

I think it would be helpful if
- drop_channels would be a bit more verbose on what it does
- particularly, if it would warn if the channels intended to be dropped are not found, and if the action would result in no drops at all
- we could do a little `ch_names = [ch_names] if isinstance(ch_names, string_types) else ch_names` at the beginning of `.drop_channels`

agramfort, 2016-08-29:
+1 for a ValueError when trying to drop a channel that is not present in
the data

jona-sassenhagen, 2016-08-29:
Ok I will get on this ASAP, and afterwards, when it's still open, the ica.get_components issue.

PR #3546 (2016-08-31): MRG drop_chans error for trying to drop channels that aren't there
closes #3542

Edit: wait, what?
closed 2016-08-31 by agramfort