mne-tools/mne-python #3543: ENH: reading event header in CTF

kingjr, 2016-08-29:
It seems that Fieldtrip has the ability to read event type headers, and get strings for each event value. I don't think MNE has this functionality yet.

jona-sassenhagen, 2016-08-30:
Check out the Brainvision and EEGLAB readers for API (assuming I've understood the problem correctly)

larsoner, 2019-05-29:
@nugenta has CTF data with `MarkerFile.mrk` that looks like the following (@kingjr any memory if this is what you were talking about three years ago?):

```
PATH OF DATASET:
/some/path

NUMBER OF MARKERS:
8

CLASSGROUPID:
3
NAME:
trigger
COMMENT:
Trigger Marker corresponding to trigger (CondBitPattern=0x80, CondBitMask=0x80, Flag0to1Only=1,MinStableLength=0,TransientSearchMask=0x0)
COLOR:
red
EDITABLE:
Yes
CLASSID:
1
NUMBER OF SAMPLES:
1
LIST OF SAMPLES:
TRIAL NUMBER		TIME FROM SYNC POINT (in seconds)
                   0				                  +0

CLASSGROUPID:
+3
NAME:
rightpuff
COMMENT:
(PositionFile=/tmp/tmp8kv3_V)
COLOR:
green
EDITABLE:
Yes
CLASSID:
+2
NUMBER OF SAMPLES:
100
LIST OF SAMPLES:
TRIAL NUMBER		TIME FROM SYNC POINT (in seconds)
                   0				      +2.01333333333
                   0				      +6.51583333333
...
                   0				      +590.179166667
                   0				      +594.814166667

CLASSGROUPID:
+3
NAME:
beep
COMMENT:
(PositionFile=/tmp/tmpQLWE4n)
COLOR:
green
EDITABLE:
Yes
CLASSID:
+3
NUMBER OF SAMPLES:
90
LIST OF SAMPLES:
TRIAL NUMBER		TIME FROM SYNC POINT (in seconds)
                   0				      +3.65083333333
                   0				      +7.90083333333
...
                   0				      +585.976666667
                   0				      +588.586666667
```

I think if this file is present we should parse it and add as annotations. @massich if I send you the file can you take a look?

agramfort, 2019-05-29:
what soft produced these files? ctf console?

>

larsoner, 2019-05-29:
Yes the CTF software allows you to mark events

agramfort, 2019-05-29:
we should indeed be able to read these...
do we know how brainstorm or fieldtrip reads these?

nugenta, 2019-05-31:
The acq console creates the file, but generally only has the initial trigger mark.

Then there’s a whole host of command line helper functions that can parse the ADC channels to add marks, or use thresholds on an optical sensor channel, etc. The marks can also be manipulated manually (with a graphical interface) in the CTF DataEditor software, which is the browsing tool. So you can easily and rapidly manually clean a dataset by pulling out marks that occur during an artifact (useful if you don’t have EOG channels, or muscle artifacts).

Allison C. Nugent, PhD
Director, MEG Core Facility
NIMH/NIH/DHHS
10 Center Drive, MSC 1059
Building 10 Room 4N242
Bethesda, MD 20892-1059
301-451-8863
nugenta@nih.gov<mailto:nugenta@nih.gov>

nugenta, 2019-05-31:
Tom Holroyd, in our group, is actually working on this… he already had a way to parse them into python (with a sed backend, lol), it’s relatively trivial to put it in the format mne-python expects. He has a pyctf module just for manipulating ctf datasets, there are likely many functions that would be useful.

Allison C. Nugent, PhD
Director, MEG Core Facility
NIMH/NIH/DHHS
10 Center Drive, MSC 1059
Building 10 Room 4N242
Bethesda, MD 20892-1059
301-451-8863
nugenta@nih.gov<mailto:nugenta@nih.gov>

hyperbolicTom, 2019-05-31:
Hi. Tom Holroyd here. Yes, I have (now pure) python code that reads MarkerFile.mrk of CTF datasets.

Eric suggested using Annotations to hold them, as opposed to creating an event array. Annotations have a duration, however the marks in a CTF dataset do not, they only indicate a time point in seconds.

As Allison mentioned, there are CTF tools for converting the raw trigger events in the STIM channel to named marks. This can be done with DataEditor or typically, two programs called thresholdDetect (creates marks from ADCs or similar) and addMarker (can create marks from a file or from the STIM channel like mne.find_events). (I also have a python version of thresholdDetect.) None of these tools include the notion of a duration associated to a mark.

Often, one wishes to score a dataset based on some behavioral response or similar, and then include in subsequent analyses only those marks in "correct" trials. One may also wish to manipulate the time of marks for ease of processing, such as moving a stimulus category mark to a response mark. We also have python code for that. Thus, part of the pre-processing in our normal workflow is to threshold, filter, and mark a dataset (including removing marks in bad trials). We have many such pre-processed datasets, where the important "event" information is stored in the MarkerFile.mrk file inside the dataset.

It would be easy to create an Annotations instance directly from the MarkerFile.mrk file, with the duration always set to zero since it's just a collection of (time, name) pairs. It would also be easy to create a simple interface such as

    event, event_id = raw.annotations.get_marks(['marker1', 'marker2', ...])

where event is the usual event array and event_id is a dict like `{'marker1': 1, 'marker2': 2}`, both suitable for passing to Epochs() or whatever, and where the marker names are read directly from the CTF dataset.

One issue: MNE concatenates the CTF "trials", so that while CTF style marks are actually (trial, time, name), MNE marks are (time, name). At least that's how mne.find_events() structures the event array. Is this true for annotations as well? There seems to be some notion of "absolute time" associated to an annotation. That is, if I have a trial that lasts 25 seconds, then an arbitrary delay between collection of the first trial and a second, a marker placed at the beginning of the second trial would have a CTF time of (1, 0) --- second trial, time 0 --- and find_events() converts that to a sample number of 25 * sfreq. Do Annotations work the same way? I'm a little confused by the "meas_date" and "orig_time" things, especially in the case of non-contiguous data. If when a CTF dataset is opened we did something like this

    raw.set_annotations(mne.Annotations(..., orig_time = 0)

would that use the same time structure as find_events()? Then getting whichever marks you want by name would be easy.

larsoner, 2019-05-31:
> Often, one wishes to score a dataset based on some behavioral response or similar, and then include in subsequent analyses only those marks in "correct" trials... We have many such pre-processed datasets, where the important "event" information is stored in the MarkerFile.mrk file inside the dataset.

These sorts of things can end up living in Annotations, so it seems like a good place to put it.

> I'm a little confused by the "meas_date" and "orig_time" things, especially in the case of non-contiguous data. If when a CTF dataset is opened we did something like this

For non-contiguous raw CTF data (?), I don't think we insert any "gaps" between the intervals, just concatenate them together. But you should be able to see this with `raw.plot` -- if there are just vertical red lines to indicate gaps, we don't do any zero insertion -- if there are periods of zeros that are red, then we do. (Neuromag has the option to pause and restart recordings, and for these files there are these zeroed-out sections.)

In either case, some sort of running offset should allow you to get the `raw.annotations` set properly for each trial. Maybe try implementing it with Annotations and see if the events line up at the correct times?

hyperbolicTom, 2019-05-31:
So it's OK to have an annotation with a zero duration?

Yeah, CTF datasets can be non-contiguous. You pause recording, then restart (a new "trial" in CTF speak, although trials can be and often are contiguous). The AUX.ds, btw, is a contiguous copy that records through the breaks for those who want that, but it's a larger file).

Sure, I'll try it and see what happens. :-)

larsoner, 2019-05-31:
> So it's OK to have an annotation with a zero duration?

Yes I think currently we do have some like this, such as `EDGE_boundary` that actually give the boundary locations when you concatenate raw instances.

In principle if no zeros are inserted in the CTF recordings, we should also have the `EDGE_boundary` annotations added, too. But that should be a separate PR. (Or maybe we do it already, I haven't checked.)

agramfort, 2019-06-02:
+1 to use an Annotations

the way I would suggest to proceed:

- create mne.io.ctf._read_annotations_ctf('fname.mrk') files
- update
https://github.com/mne-tools/mne-python/blob/master/mne/annotations.py#L551 to
it can read the ctf .mrk files based on extension
- make sure the mne.io.read_raw_ctf reads these annotations automatically
without creating a stim channel

note that I find .mrk files in all the ctf files we share (bst_phantom_ctf,
bst_auditory, MNE-spm-face)
and so I think we just never bothered reading them in the read_raw_ctf
function.

massich, 2019-06-04:
(I've no idea why I didn't see this before) But yes, @larsoner sent me the files I'll take a look.

massich, 2019-06-04:
we should definetly add `ctf` to supported annotations files.

agramfort, 2019-06-04:
@massich you have such files

alex@:~$ cd ~/mne_data/
alex@:mne_data$ find . -name "*.mrk"
./MNE-brainstorm-data/bst_auditory/MEG/bst_auditory/S01_AEF_20131218_01.ds/MarkerFile.mrk
./MNE-brainstorm-data/bst_auditory/MEG/bst_auditory/S01_AEF_20131218_02.ds/MarkerFile.mrk
./MNE-brainstorm-data/bst_phantom_ctf/emptyroom_20150709_01.ds/MarkerFile.mrk
./MNE-brainstorm-data/bst_phantom_ctf/phantom_200uA_20150709_01.ds/MarkerFile.mrk
./MNE-brainstorm-data/bst_phantom_ctf/phantom_20uA_20150603_03.ds/MarkerFile.mrk
./MNE-brainstorm-data/bst_raw/MEG/bst_raw/subj001_somatosensory_20111109_01_AUX-f.ds/MarkerFile.mrk
./MNE-fieldtrip_cmc-data/SubjectCMC.ds/MarkerFile.mrk
./MNE-phantom-4DBTi/1/e,rfhp1.0Hz.mrk
./MNE-spm-face/MEG/spm/SPM_CTF_MEG_example_faces1_3D.ds/MarkerFile.mrk
./MNE-spm-face/MEG/spm/SPM_CTF_MEG_example_faces2_3D.ds/MarkerFile.mrk

PR #6591 (2019-08-01): ENH: read CTF event header as annoations
fixes #3543
closed 2019-08-01 by agramfort