mne-tools/mne-python #3550: gat.to_evoked() ?

jona-sassenhagen, 2016-08-30:
How about a `GeneralizationAcrossTime.to_evoked` method that returns an evoked array for the score time series, with either only the diagonal, or one "channel" for each training time point? That way, some of the plotting methods for EvokedArray could be leveraged (e.g., `plot_compare_evokeds` ...).

Just a thought.

@kingjr

kingjr, 2016-08-31:
This is related to #3424. 

IMO it's a good idea, but I wouldn't implement it on the GAT directly.

We're trying to make the supervised API consistent, and there's a big chance that we'll favour the dev of sklearn-style estimators taking X and y as opposed to hybrid objects that take epochs, apply the cv internally, deal with the predict_method etc.

However, a `to_evoked` function/method may be useful to extract the `patterns_` from all spatial filter estimators (Xdawn, CSP, TimeDecoding, UnsupervisedSpatialFilter). 

This would require making a `SpatialFilterMixin` that computes the patterns from the filters for all of these classes. Would you be up for it?

jona-sassenhagen, 2016-08-31:
>Would you be up for it?

Hm... Probably not any time soon. I'm still fairly hopeless when it comes to classes.

kingjr, 2016-08-31:
I just gave it a try, but rapidly hurt some API issue. Here are some additional thoughts

### SpatialFilterMixin

Create a mixin class for spatial filter estimators (i.e. Xdawn, CSP, SearchLight & GAT) and init all of these estimators with a `compute_patterns` bool param.

This makes sense in term of code quality, however, Xdawn and CSP already have their own computation of the patterns (that's a small problem)

### The SearchLight case

If we init SearchLight with a `compute_pattern` bool param, it is unclear to me
1) how we will deal with SearchLight that are not across time, but e.g. across frequencies.
2) Whether we make it smart to retrieve the coef of pipeline estimators (e.g. StandardScaler() + LogisticRegression()) by looping the `steps`, or just look for a `coef` attributes.
3) Whether we have to invert_transform the coefs with the preceeding steps. E.g. StandardScaling should be used to inv transform the coef to be back a physical unit.
4) If we apply it to the generic `SearchLight` directly, we won't be able to compute patterns if `X` is not 3 dimensional. E.g. if you do a time freq decomposition before doing a searchlight, each estimator will be based on 3 dimension, and we can't compute a covariance anymore.

### Covariance parameterizing

Do we want to parametrize the covariance computation (e.g. adding a regularlization) before computing the pattern. I think not, but I'm not sure.

jona-sassenhagen, 2016-09-01:
Note, I wasn't actually thinking about recovering the spatial patterns, just the decoder score time series.

Though somehow automatically recovering the patterns would be very cool.

kingjr, 2016-09-01:
The coefs can be found with a one liner something like `[est.coef_ for fold in gat.estimators_ for est in folds]` so I don't think it's a high priority, especially as the filters themselves are not easily interpretable.

jona-sassenhagen, 2016-09-01:
Yeah, I meant recovering some form of interpretable pattern (e.g., via the Haufe trick).

kingjr, 2016-09-01:
But then we need to compute the covariance. I'm *think* that, strictly speaking, we need to compute a covariance within the CV no?

jona-sassenhagen, 2016-09-01:
Probably. But IIRC neither you nor Denis care much for the Haufe trick, so ...

kingjr, 2016-09-01:
No, I do care, I think it's one of the important missing features that should be present for all spatial filters. It is mainly a dev issue.

PR #3949 (2017-02-21): MRG+1: get patterns coefs with example
After a couple of iterations, I believe the best way to deal with the extraction and plotting of spatial filters and spatial patterns is to i) ensure that we use the LinearModel class at the end of our pipeline and ii) use a function that can extract and inverse transform these coefficients.

This API allows a full sklearn compatibility, and avoid adding additional complex attributes to `SearchLight` and `GeneralizationLight` (filters_, inverse_filters, inverse_patterns etc)

```python
clf = make_pipeline(Vectorizer(), StandardScaler(), LinearModel(Logist [... cut]
closed 2017-02-21 by kingjr