mne-tools/mne-python #3554: find_events fails on TRIUX data with legacy stim channels

jjnurminen, 2016-09-01:
Not sure if this is a bug or not, but I think the situation could be improved.

Typically Elekta TRIUX (and recent Vectorview?) systems record triggers only on the digitally coded channel STI101, which contains all trigger information. However, in the data acquisition, there is a possibility of also acquiring 'legacy' trigger channels STI001, STI002, etc. These are binary channels which indicate the corresponding trigger channel being on/off. The main reason for acquiring these would be to support legacy software that does not support STI101.

The problem is that `mne.find_events` fails for raw files that contain both STI101 and the legacy channels, because it regards all of them as stim channels and considers them simultaneously. Setting `stim_channel=STI101` remedies the situation. I think an easy solution would be to always use STI101 if it exists, and disregard the legacy channels.

(In some setups, the trigger channel STI201 is also used separately, enabling 32 trigger lines. However this is kind of a rare us case, so I think it doesn't need to be considered by default. Usually STI101 and STI201 are mirrored, i.e. contain identical information.)

jjnurminen, 2016-09-01:
Actually this is nicely remedied by returning STI101 in `_get_stim_channel()`
Will submit a PR.

Eric89GXL, 2016-09-01:
Yeah it should try `STI 014`, `STI101`, then whatever it does now.

jjnurminen, 2016-09-01:
I already wrote a small patch but it ended up into the wrong branch and my
attempt at cherry-picking failed. (insert tearful emoji)

Eric89GXL, 2016-09-01:
Nothing wrong with a little bit of manual cut and paste in desperate times

PR #3560 (2016-09-02): default to STI101 instead of STI 014
Newer Elekta systems (at least TRIUX) use 'STI101' as the combination trigger channel, so it should be the default. If not found, fall back to 'STI 014'
Fixes #3554
closed 2016-09-02 by Eric89GXL