mne-tools/mne-python #3555: eeglab reader warns that it failed to find events when no events are present in the file

mmagnuski, 2016-09-01:
I either have no events in the file (for example 'resting-state' activity) or have the events somewhere else and I would like to turn off the warnings that I get from eeglab reader. I tried changing `mne.set_log_level()` as well as setting `warnings.filterwarnings("ignore")` but I still get the warnings.
The warning is [here](https://github.com/mne-tools/mne-python/blob/master/mne/io/eeglab/eeglab.py#L613).
I don't think the reader should warn about not being able to find events when there are no events to be found. The intent of this error was to signal when no events could be coerced to integers etc.

jasmainak, 2016-09-01:
maybe @jona-sassenhagen has comments?

Eric89GXL, 2016-09-01:
> I don't think the reader should warn about not being able to find events when there are no events to be found.

Agreed, can you open an issue to only warn if the file has events in it?

jona-sassenhagen, 2016-09-01:
Im on it.

jona-sassenhagen, 2016-09-01:
That'd be just making L612 in io/eeglab/eeglab.py

    if len(events) < 1 and len(types) > 0:

right?

mmagnuski, 2016-09-01:
@jona-sassenhagen yes, that should work (+ a test maybe? :) )

jona-sassenhagen, 2016-09-01:
For a test, we'd have to write a new EEGLAB structure. So we're looking at a few lines of test for half a line of code.

@Eric89GXL ?

Eric89GXL, 2016-09-01:
We can live without a test, just don't tell @agramfort

mmagnuski, 2016-09-01:
:no_mouth: :speak_no_evil:

agramfort, 2016-09-01:
too bad for you I still have time to read emails...

but I can live here without a test...

PR #3557 (2016-09-05): MRG: eeglab io: don't warn that no events can be converted if 0 events found
closes #3555

@mmagnuski
commit c4174f7f (2016-09-05): MRG: eeglab io: don't warn that no events can be converted if 0 events found (#3557)

* eeglab io: don't warn that no events can be converted if if zero events found

closes #3555

* Update eeglab.py

* Update eeglab.py

* fix test
closed 2016-09-05 by agramfort