mne-tools/mne-python #3558: read_montage() broken for sfp file

sstober, 2016-09-01:
Hello,

it seems like commit 4ff58df47075b1c1edbc933d57db9eca17c38a61 has broken read_montage() for an sfp file that caused no problems before. The issue seems to be the line breaks. During debug, I could see that the whole file content ends up in line 0. Looks like split("\n") does not work. Nano is telling me that the file is in "Mac format". This might be the reason.

Cheers,
Sebastian

jona-sassenhagen, 2016-09-01:
Argh ... this should be my responsibility, but I'm terrible with encoding and regexp and stuff. Can anybody else take a look?

jona-sassenhagen, 2016-09-01:
If nobody has the time, could you upload the problematic file somewhere?

sstober, 2016-09-01:
Here's a dropbox link:
https://www.dropbox.com/s/3nmmx802aiwo0vm/bad.sfp?dl=0

On 01/09/16 20:04, jona-sassenhagen wrote:
> If nobody has the time, could you upload the problematic file somewhere?
>

closed 2016-09-02 by agramfort