mne-tools/mne-python #3562: Handling of short events is inappropriate for combination trigger channels

jjnurminen, 2016-09-02:
I noticed that `find_events()` fails when there are overlapping triggers, which can (and often do) manifest as arbitrarily short events on a combination trigger channel (e.g. 'STI101' on TRIUX systems), at least when using `consecutive=True`.

E.g. let's say that there is trigger pulse on channel 1 and one millisecond later a pulse on channel 2. On the combination channel, this results in 'spurious' event duration of 1 ms, even though there is nothing wrong with the data.

This is remedied by setting `shortest_event=0`, but I think checking the event duration does not make sense for combo channels.

Eric89GXL, 2016-09-02:
It probably doesn't make sense in cases where binary channel 1 and binary channel 2 are used independently, but this was done for another use case, e.g. where a user typically wants to trigger a decimal "3" but the hardware isn't perfect so can have a 1 followed almost immediately by a 2 (summing to 3). So it's likely to break some people's setups if we change this.

I think we'll have to live with just making a documentation change about it :(

FWIW in our lab we always use the trigger lines separately, and I actually use `find_events` on `STI001`, `STI002`, etc. to be explicit about these things.

jjnurminen, 2016-09-02:
>this was done for another use case, e.g. where a user typically wants to trigger a decimal "3" but the >hardware isn't perfect so can have a 1 followed almost immediately by a 2 (summing to 3)

Having several channels trigger simultaneously to indicate e.g. a decimal 3 is a very common use case, but I never had any problems with it (using either MEG internal triggering, Presentation or Stim2). I think the DACQ event detection (which I have been using) allows some small timing mismatch between triggers, which may explain why it worked.

Now I'm not 100% sure whether find_events can be made to work in the same fashion. I have to do some tests. This issue is of course related to the Elekta averager.

jjnurminen, 2016-09-02:
The Elekta manual states: 
"To avoid misinterpretations of the trigger combinations due
to race conditions, the following procedure is implemented in
the on-line averager. Assume sample k with all trigger lines
inactive followed by sample k + 1 having one or more active
trigger lines. Sample k + 1 is used as the timing reference, but
the trigger line combination is determined from sample k + 2.
This allows a slight asynchrony between the on-sets of the
trigger lines."

dgwakeman, 2016-09-02:
Yes, this is why I added the `shortest_event=2` . Interestingly, this offers a slight difference between the exact start of an event that our find_events will use and Elekta will; however, this effect should be minimal. IIRC to change the model to work like the Elekta code would require a significant re-write (and would probably result in what would amount to an API change).

The race condition occurs on both Neuromag systems, I have used. It can be seen in my dataset:
http://www.nature.com/articles/sdata20151

jjnurminen, 2016-09-02:
Thanks @dgwakeman. However, with my test data I got ValueError from `find_events` (due to short events), while the Elekta averager found a lot epochs. After setting shortest_event=0, I think I got the same result as Elekta (need to verify this next week).

So I'm a bit confused: according to the above, shouldn't the Elekta code also disregard single-sample events (since it looks one sample forward, where the single-sample trigger has already disappeared?)

Eric89GXL, 2016-09-02:
@dgwakeman sorry for my ignorance, I didn't know that dataset existed. Any interest in writing an MNE-Python tutorial (and adding it to `datasets` somehow)? If not, do you think it's worth someone else doing it to show how to process the data with MNE-Python? I don't think we have anything about integration with fMRI currently.

> according to the above, shouldn't the Elekta code also disregard single-sample events 

Yes that's what it sounds like from the documentation, but it might not be 100% correct. The way I would check is by actually passing it data with known values to see what it does. It's possible they added a one-sample exception without updating the docs, for example.

jjnurminen, 2016-09-02:
> check is by actually passing it data with known values to see what it does.

I'll try to do some checks. It's interesting that different parts of the Elekta software also interpret triggers differently, at least in 'pathological' cases. For example, the ElektaAverager test data causes a mismatch between the online averager and MaxAve.

>The race condition occurs on both Neuromag systems, I have used.

@dgwakeman Can you clarify what you mean by race condition here?

dgwakeman, 2016-09-02:
@jjnurminen, I suspect (read don't know) it depends on the actual events in the trial. The case is really specific, so it is possible it only deals with one type and not another.

e.g.
Real event is 3 for 10ms on 1khz data
STI101:

13333333333

I don't know what happens if you have this
Real event is 1 for 1ms then an event 2 for 1ms
12000000000
Events 2 and 1 are on separate inputs so the code may know this and exclude it from using the k+1 value

dgwakeman, 2016-09-02:
@Eric89GXL this is something @agramfort and @jasmainak are working on. Actually, all the software packages are making tutorials using this dataset. I was supposed to go to a satellite at biomag on it, but unfortunately, given my unemployed status, I cant arrange the funds.

dgwakeman, 2016-09-02:
@jjnurminen It all comes down to the binary input sum thing. That is where the race condition is. STI001 and STI002 are separate input channels to the NEUROMAG system, I think it has to do with something in how the hardware/firmware read the STI channels and synthesizes them into STI101, because this problem is virtually independent of sampling frequency i.e. I don't see cases of 11333... when I switch from acquisition at 1k vs. 2k

jjnurminen, 2016-09-09:
Just FYI, I now have some test data from a new Stim2 system which includes a gadget called StimTracker. This box tracks actual physical appearance of stimuli using photodetectors etc. to produce accurate onset triggers. When setting up averaging categories, one then must define the onset trigger as timing reference and then a stimulus category specific trigger as a conditional event. Since the onset and the category triggers happen quite close to each other and overlap in time, this setup easily causes problematic behavior with `find_events`, i.e. excessively short events are detected and ValueError is raised.
However, the good news is that at least with this test data, setting `shortest_event=1` perfectly replicates the results from Elekta's own averager.

jjnurminen, 2016-09-09:
So the conclusion is like this:

- `shortest_event=2`(the default) may cause an exception on combination channels (e.g. STI101) due to closely overlapping triggers; in many cases, this may be solvable by masking trigger bits away
- Using `shortest_event=1` will fix the problem, but it will cause slightly asynchronous trigger onsets to be detected as separate events, which perhaps was not intended by the user

So I think this one can be closed. Thanks to @andreyzhd for an enlightening discussion about the issue.

Eric89GXL, 2016-09-09:
Thanks for following up @jjnurminen and @andreyzhd

closed 2016-09-09 by Eric89GXL