mne-tools/mne-python #3567: New filter defaults

jaeilepp, 2016-09-06:
I tried the new defaults when analysing the multi-modal faces dataset. I filtered the data with new filter defaults
```
raw.filter(1, 40, method='iir', n_jobs=1, h_trans_bandwidth='auto', l_trans_bandwidth='auto', verbose='INFO')
```
and was surprised to see the 50 Hz line noise in the results with cut-off frequency of 40 Hz.
[image]
[image]

(For some reason the bad channel isn't as visible in the picture as in the mpl figure.)

Eric89GXL, 2016-09-06:
IIR defaults shouldn't have changed, just FIR

Eric89GXL, 2016-09-06:
The trans_* arguments aren't used with IIR actually

On Sep 6, 2016 6:45 AM, "Eric Larson" <larson.eric.d@gmail.com> wrote:

> IIR defaults shouldn't have changed, just FIR
>

jaeilepp, 2016-09-06:
Ok, I see. Closing this.

closed 2016-09-06 by jaeilepp