mne-tools/mne-python #3571: Need robustness to uncoerceable filter cutoffs in BV headers

jona-sassenhagen, 2016-09-07:
I think a recent PR leads to problems with the BV importer if the header file has a string for the filter cutoff.

```
raw = mne.io.brainvision.read_raw_brainvision(fname)

Online software filter detected. Using software filter settings and ignoring hardware values
---------------------------------------------------------------------------
ValueError                                Traceback (most recent call last)
<ipython-input-4-8884b0022b63> in <module>()
      1 fname = "/Users/jona/Downloads/odd_00.vhdr"
----> 2 raw, events = get_raw(fname)

<ipython-input-3-28df6c2be3b1> in get_raw(fname)
      3 
      4 def get_raw(fname):
----> 5     with mne.io.brainvision.read_raw_brainvision(fname, montage=montage, 
      6 #                                                 preload=True
      7                                                 ) as raw:

/Users/jona/tools/mne-python/mne/io/brainvision/brainvision.py in read_raw_brainvision(vhdr_fname, montage, eog, misc, scale, preload, response_trig_shift, event_id, verbose)
    616                           misc=misc, scale=scale, preload=preload,
    617                           response_trig_shift=response_trig_shift,
--> 618                           event_id=event_id, verbose=verbose)

/Users/jona/tools/mne-python/mne/io/brainvision/brainvision.py in __init__(self, vhdr_fname, montage, eog, misc, scale, preload, response_trig_shift, event_id, verbose)

/Users/jona/tools/mne-python/mne/utils.py in verbose(function, *args, **kwargs)
    691         with use_log_level(verbose_level):
    692             return function(*args, **kwargs)
--> 693     return function(*args, **kwargs)
    694 
    695 

/Users/jona/tools/mne-python/mne/io/brainvision/brainvision.py in __init__(self, vhdr_fname, montage, eog, misc, scale, preload, response_trig_shift, event_id, verbose)
     80         vhdr_fname = os.path.abspath(vhdr_fname)
     81         info, fmt, self._order, mrk_fname, montage = _get_vhdr_info(
---> 82             vhdr_fname, eog, misc, scale, montage)
     83         events = _read_vmrk_events(mrk_fname, event_id, response_trig_shift)
     84         _check_update_montage(info, montage)

/Users/jona/tools/mne-python/mne/io/brainvision/brainvision.py in _get_vhdr_info(vhdr_fname, eog, misc, scale, montage)
    498         else:
    499             if hp_s:
--> 500                 info['highpass'] = np.min(np.array(highpass, dtype=np.float))
    501                 info['highpass'] = 1. / info['highpass']
    502             else:

ValueError: could not convert string to float: 'Off'
```

The end of the .vhdr looks like this:

```
S o f t w a r e  F i l t e r s
==============================
#     Low Cutoff [s]   High Cutoff [Hz]   Notch [Hz]
...
63     Off              Off                Off
64     Off              Off                Off
```
(etc)

It worked with this file before.
I have previously worked on a similar issue, but it seems I wasn't thorough enough: #1834

@palday ?

I can in principle do this, but I'd rather work on my long list of outstanding issues.

palday, 2016-09-07:
Jepp, can do - can you post the entire header file so that I can include a test?

jona-sassenhagen, 2016-09-07:
http://pastebin.com/gXdD7qss

This was from some kind of test run IIRC.

palday, 2016-09-08:
It worked previously because software filters were ignored -- if you had something similar in your hardware filters (some `Off` but some on), it would have failed there as well because `Off` was only handled in those cases where it was the sole value.

PR #3576 (2016-09-10): MRG: BV Reader: Allow 'Off'/'DC'/'NaN' as possible value in heterogeneous filter settings
Closes  #3571. 

Currently only works for non-numeric filter cutoffs in `('NaN', 'Off', 'DC')` because those are the only ones previously supported for homogeneous filter settings and the only ones that there is an 'obvious' way to handle.

Because software filters are considered to just 'replace' hardware filters when present, this will work for heterogeneous software or hardware filters. One day, it might make sense to parse both hardware and software filters and then take the more conservative approach [... cut]
closed 2016-09-10 by agramfort