mne-tools/mne-python #3572: info.pick_info method?

jona-sassenhagen, 2016-09-07:
Any chance of adding `mne.pick_info` as a method to info? That would make it more elegant to pass the (picked) info to other functions - without creating a handle to an intermediate object.

agramfort, 2016-09-08:
type code you would like to write

jona-sassenhagen, 2016-09-08:
`mne.viz.plot_topomap(array_for_data_chans, info.copy().pick(...))`

instead of:

```
info_picked = mne.pick_info(info, ...)
mne.viz.plot_topomap(array_for_data_chans, info_picked)
```

I guess you could also do

```
from mne import pick_info
mne.viz.plot_topomap(array_for_data_chans, pick_info(info.copy(), ...))
```

But we usually allow to chain methods for objects.

agramfort, 2016-09-08:
is that

info.pick_types that you need?

jona-sassenhagen, 2016-09-08:
... closing this :3

closed 2016-09-08 by jona-sassenhagen