mne-tools/mne-python #3578: BUG: Breakages in time_gen with latest numpy and sklearn

Eric89GXL, 2016-09-08:
First one probably needs to be `np.maximum` instead of `np.max`:
```
  File "/home/larsoner/custombuilds/mne-python/mne/decoding/csp.py", line 604, in _ajd_pham
    tmp2 = (tmp * g12 - g21) / np.max(omega - 1, 1e-9)
  File "/home/larsoner/.local/lib/python2.7/site-packages/numpy-1.12.0.dev0+b91e8d8-py2.7-linux-x86_64.egg/numpy/core/fromnumeric.py", line 2249, in amax
    return amax(axis=axis, out=out, **kwargs)
  File "/home/larsoner/.local/lib/python2.7/site-packages/numpy-1.12.0.dev0+b91e8d8-py2.7-linux-x86_64.egg/numpy/core/_methods.py", line 26, in _amax
    return umr_maximum(a, axis, None, out, keepdims)
TypeError: 'float' object cannot be interpreted as an index
```
Second one must be due to sklearn change:
```
  File "/home/larsoner/custombuilds/mne-python/mne/decoding/base.py", line 670, in _set_cv
    cv = XFold(n_folds=cv)
TypeError: __init__() got an unexpected keyword argument 'n_folds'
```
@kingjr do you have time to look?

agramfort, 2016-09-08:
see https://github.com/mne-tools/mne-python/pull/3579

let's see if CIs are happy...

kingjr, 2016-09-08:
I actually just hit that error, if alex s fix doesn't pass I'll have a look. Locally changing n_folds to n_splits solved it; didn't touch the np.max however

Eric89GXL, 2016-09-08:
Pretty sure the `np.max` is a bug, can you submit a separate fix?

PR #3580 (2016-09-09): MRG sklearn .18 n_splits
attempt to close #3578 & #3579
closed 2016-09-08 by kingjr