mne-tools/mne-python #3582: RandomizedPCA deprecation

Eric89GXL, 2016-09-09:
I get this deprecation warning:
```
DeprecationWarning: Class RandomizedPCA is deprecated; RandomizedPCA was deprecated in 0.18 and will be removed in 0.20. Use PCA(svd_solver='randomized') instead. The new implementation DOES NOT store whiten components_. Apply transform to get them.
```
It would be good to make this change. I can do it unless you're motivated @kingjr

Eric89GXL, 2016-09-09:
Likely related, but might expose a bug here or in sklearn:
```
======================================================================
ERROR: Test that EOG channel is included when performing ICA
----------------------------------------------------------------------
Traceback (most recent call last):
  File "/home/larsoner/custombuilds/nose/nose/case.py", line 197, in runTest
    self.test(*self.arg)
  File "/home/larsoner/custombuilds/mne-python/mne/utils.py", line 812, in dec
    return function(*args, **kwargs)
  File "/home/larsoner/custombuilds/mne-python/mne/preprocessing/tests/test_ica.py", line 680, in test_eog_channel
    ica.fit(inst, picks=picks1)
  File "<string>", line 2, in fit
  File "/home/larsoner/custombuilds/mne-python/mne/utils.py", line 693, in verbose
    return function(*args, **kwargs)
  File "/home/larsoner/custombuilds/mne-python/mne/preprocessing/ica.py", line 357, in fit
    tstep, verbose)
  File "/home/larsoner/custombuilds/mne-python/mne/preprocessing/ica.py", line 423, in _fit_raw
    self._fit(data, self.max_pca_components, 'raw')
  File "/home/larsoner/custombuilds/mne-python/mne/preprocessing/ica.py", line 526, in _fit
    data = pca.fit_transform(data.T)
  File "/home/larsoner/.local/lib/python2.7/site-packages/sklearn/decomposition/pca.py", line 312, in fit_transform
    U, S, V = self._fit(X)
  File "/home/larsoner/.local/lib/python2.7/site-packages/sklearn/decomposition/pca.py", line 351, in _fit
    return self._fit_truncated(X, n_components, svd_solver)
  File "/home/larsoner/.local/lib/python2.7/site-packages/sklearn/decomposition/pca.py", line 420, in _fit_truncated
    % (n_components, n_features, svd_solver))
ValueError: n_components=5 must be between 1 and n_features=4 with svd_solver='randomized'
```

kingjr, 2016-09-09:
It s not in a supervised analysis is it? I think it s out of my scope.

PR #3587 (2016-09-09): FIX: random pca solver in sklearn v.18
closes #3582
closed 2016-09-09 by Eric89GXL