mne-tools/mne-python #3583: BUG: CSP transform doesn't comply with sklearn API

kingjr, 2016-09-09:
`CSP.transform` currently returns the CSP features averaged over time. 

IMO this is not a transform in the sklearn sense, it's a feature extraction. The transform should be the time course of each CSP i.e. return an array of `shape (n_epochs, n_filters, n_times)`.

Typically, if I want to do a dim reduction by CSP e.g. `make_pipeline(CSP(), TimeFrequency())` the current API can't work.

I don't think there's a way to do a nice deprecation cycle for this :/, except by redoing another object like we did for the Xdawn...

cc @alexandrebarachant @agramfort

agramfort, 2016-09-09:
I would add a parameter to CSP to average over time or not.

the averaging is the standard use of CSP

kingjr, 2016-09-09:
ok

PR #3586 (2016-09-13): MRG: add param to CSP to transform into average power or time course
closes #3583 

cc @alexandrebarachant
closed 2016-09-13 by agramfort