mne-tools/mne-python #3588: BUG: Failing example on master

Eric89GXL, 2016-09-09:
CircleCI is angry because of this:
```
../examples/decoding/plot_decoding_spatio_temporal_source.py failed to execute correctly:Traceback (most recent call last):
  File "/home/ubuntu/mne-python/sphinx-gallery/sphinx_gallery/gen_rst.py", line 518, in execute_code_block
    exec(code_block, example_globals)
  File "<string>", line 65, in <module>
  File "/home/ubuntu/mne-python/mne/source_estimate.py", line 1334, in plot
    time_unit=time_unit)
  File "/home/ubuntu/mne-python/mne/viz/_3d.py", line 804, in plot_source_estimates
    colorbar=colorbar, **kwargs)
  File "/home/ubuntu/miniconda/envs/circleenv/lib/python2.7/site-packages/surfer/viz.py", line 830, in add_data
    min = array.min()
  File "/home/ubuntu/miniconda/envs/circleenv/lib/python2.7/site-packages/numpy/core/_methods.py", line 29, in _amin
    return umr_minimum(a, axis, None, out, keepdims)
ValueError: zero-size array to reduction operation minimum which has no identity
```
Anybody seeing this locally? Doesn't happen for me but I haven't tried my Conda envs yet.

jaeilepp, 2016-09-12:
Where you see this failure? Master seems to be fine atm https://circleci.com/gh/mne-tools/mne-python/2503

Eric89GXL, 2016-09-12:
Every PR was failing for a number of days. Maybe it's fixed now, I'll check and close if that's the case.

Eric89GXL, 2016-09-12:
Still failing:

https://circleci.com/gh/mne-tools/mne-python/2516

So maybe it's just my build fork that's being problematic...?

Eric89GXL, 2016-09-14:
This is a PySurfer problem, fix at https://github.com/nipy/PySurfer/pull/163

closed 2016-09-14 by Eric89GXL