mne-tools/mne-python #3598: ENH: Add minimum-phase filtering

Eric89GXL, 2016-09-16:
We should have minimum phase filtering. It could especially be useful in realtime applications, cc @pushtheworldllc @m-brantheim. We need the upstream SciPy PR https://github.com/scipy/scipy/pull/6497 to land first. But it should only be 1-2 months away.

agramfort, 2016-09-17:
can you clarify why this would be useful for realtime?

Eric89GXL, 2016-09-17:
For things like online BCI, reducing delay is important. In linear phase
(or zero phase by delay correction) filtering the group delay in the
passband is substantially larger than in minimum phase

Eric89GXL, 2016-09-17:
(Well the actual group delay is zero for the zero phase case, but it is
acausal so you do need to wait half the filter length before doing the
delay correction -- hence an effective delay when actually trying to use it)

PR #3780 (2016-11-27): MRG: Add minimum-phase filtering
Closes #3598.
closed 2016-11-27 by Eric89GXL