mne-tools/mne-python #3599: ENH to hierarchical tags

jasmainak, 2016-09-16:
Currently, if we have event_ids which look like this:

```py
event_id = {'auditory/left': 1, 'auditory/right': 2,
            'visual/left': 3, 'visual/right': 4}
```

it is possible to do:

```py
epochs[('auditory/left', 'visual/left')]
```

and 
```py
epochs['auditory'] 
```

but not:

```py
epochs[('auditory', 'visual')]
```
It seems that intuitively this should be made possible. It's also how `read_selection` works for example. Currently, I'm using `mne.concatenate_epochs` for this purpose but it's a bit annoying to use especially when you are prototyping things quickly. Thoughts?

dengemann, 2016-09-17:
Sounds good. But I was almost sure that this works. Did you try list
instead of tuple syntax?

jasmainak, 2016-09-17:
You mean ``epochs[['auditory', 'visual']]`` ? Nopes, this doesn't work either

jasmainak, 2016-09-17:
Here's a small script to reproduce:

```py
import mne
from mne.datasets import sample

data_path = sample.data_path()

raw_fname = data_path + '/MEG/sample/sample_audvis_filt-0-40_raw.fif'
event_fname = data_path + '/MEG/sample/sample_audvis_filt-0-40_raw-eve.fif'
tmin, tmax = -0.1, 0.3
event_id = {'auditory/left': 1, 'auditory/right': 2,
            'visual/left': 3, 'visual/right': 4}

raw = mne.io.read_raw_fif(raw_fname, preload=True)
events = mne.read_events(event_fname)

picks = mne.pick_types(raw.info, meg=False, eeg=True, stim=False, eog=False,
                       exclude='bads')

epochs = mne.Epochs(raw, events, event_id, tmin, tmax, proj=False,
                    picks=picks, baseline=None, preload=True,
                    add_eeg_ref=False, verbose=False)
epochs[('auditory/left', 'visual/left')]  # this works
epochs['auditory']  # this too works
epochs[('auditory', 'visual')]  # but mysteriously this doesn't work
```

Eric89GXL, 2016-09-17:
I agree that should work. I could try to fix it Monday for release.

agramfort, 2016-09-17:
+1 to make this work

jasmainak, 2016-09-17:
> I agree that should work. I could try to fix it Monday for release.

That would be really cool. It might also help for our Biomag demo

PR #3603 (2016-09-26): MRG: Better logic for epoch selection
@jona-sassenhagen do you have time to take a look? This deals with the hierarchical event id stuff and should make it a bit more consistent.

Closes #3599.
Closes #2875.
closed 2016-09-26 by agramfort