mne-tools/mne-python #3601: [Question] How to import CSV data?

ppasler, 2016-09-19:
Hi guys,

as the mailing-list link on your page does not work, I use the issue tracker for asking, how to import CSV data? 

Background:
I am using the [Emotic EPOC+](https://www.emotiv.com/epoc/) EEG headset with 14 channels (which is far  away from any medical EEG - I know). Nevertheless I wonder if there is an option to import the recorded channels which are written in simple CSV format.
`F3	F4	P7	FC6	F7	F8	T7	P8	FC5	AF4	T8	O2	O1	AF3`
`-149	-125	-128	-259	312	-138	-113	-146	-132	-133	-101	-152	-170	-225`

Subsequent question: does mne-python work with 14 channels at all?

Thanks for answering, even it is the wrong channel.
Greetz paul

dengemann, 2016-09-19:
Checkout the pandas library and it's io tools. `pd.read_csv`. From there
you can then create epochs or raw objects as described here:
http://martinos.org/mne/stable/auto_tutorials/plot_creating_data_structures.html

And to your second questions. Not sure why it shouldn't. It depends of
course on what is your goal :)

ppasler, 2016-09-19:
Hi Dennis,

thanks for the fast reply.
I will check out the link you posted (I kinda missed that one...) and see if this fits my needs.

If you are interested:
My research project at university is about [detecting driver drowsines](https://github.com/ppasler/current-adas/tree/master/project) with an EEG. I recorded some data in our driving simulator and assumed that the drowsiness increases over time (which the probands comfirmed). I then wrote a litte application and did the processing by myself (eventhough I am not very experinced with EEG and signal processing). The result was promising, but not as good as I aspected. So I want to try mne-python to see if I made some mistakes in my calculations and eventually can improve the detection rate.

ppasler, 2016-09-19:
Works like a charm, thanks!

closed 2016-09-19 by ppasler

dengemann, 2016-09-19:
Great and glad to hear! Your project sounds interesting, I hope mne-python
will be of help for you!