mne-tools/mne-python #3604: Progress Bar in GAT

teonbrooks, 2016-09-19:
Is there a place in the gat code to add a progress bar like the spatiotemporal test has? I'm running it and I can't tell if it is a taxing on the computer or if it just silently died. Even if it is an update on each fold would be sufficient.

dengemann, 2016-09-19:
Teon is back!!! :)

kingjr, 2016-09-19:
There may be a way but I don't know it.

I actually would like this feature too, as I get a tonne of useless outputs from joblib instead :/

teonbrooks, 2016-09-20:
> Teon is back!!! :)

not yet... :/ almost!

jasmainak, 2016-09-20:
we have a progressbar in utils.py. That could be used ...

kingjr, 2016-09-20:
Iirc The problem was not the problem bar itself but it's use in a
parallelized context.

Eric89GXL, 2016-09-20:
The clustering code is parallel, so we can hopefully do the same thing in
both places.

kingjr, 2016-09-20:
Right, we just have to copy 
```
    def get_progress_bar(seeds):
        # make sure the progress bar adds to up 100% across n jobs
        return (ProgressBar(len(seeds), spinner=True) if
                logger.level <= logging.INFO else None)
```

in `_GeneralizationAcrossTime` `TimeDecoding` and `SearchLight`. @teonbrooks you up for that?

teonbrooks, 2016-09-21:
i'm grounded from coding for about one more month.

jona-sassenhagen, 2016-09-24:
Yes, this would be very nice to have.

Also for Time-Freq computations.

closed 2017-07-22 by agramfort