mne-tools/mne-python #3608: BUG: the craziest we've encountered in our lifetime

teonbrooks, 2016-09-21:
We've encountered a *challenging* bug which only happens for the GAT with a Ridge + n_jobs != 1 + an info with projection.

We checked that it works well with n_jobs = 1, clf=LogisticRegression, and a homemade info with success.

The info file can be downloaded from [here](https://www.dropbox.com/s/vvfvgetr8b42hi7/info.fif?dl=0)

```python
import numpy as np
from sklearn.linear_model import Ridge
from mne import EpochsArray, create_info
from mne.io import read_info
from mne.decoding import GeneralizationAcrossTime

n_epochs, n_channels, n_times = 557, 208, 60
X = np.random.rand(n_epochs, n_channels, n_times)
y = np.random.rand(n_epochs)

info = create_info(n_channels, 500, 'mag')
our_info = read_info('/Users/teon/Desktop/info.fif')
epochs = EpochsArray(X, info)

gat = GeneralizationAcrossTime(n_jobs=-1, clf=Ridge())
gat.fit(epochs, y)

EpochsArray(X, our_info)

gat.fit(epochs, y)  # joblib hangs forever
```

joblib: 0.9.4
numpy: 1.11.1
sklearn: 0.16.1
mne: dev
JR & Teon

agramfort, 2016-09-22:
is it on a mac without mkl using python 2.7?

kingjr, 2016-09-22:
Mac and python 2.7 yes, I haven't checked mkl

Eric89GXL, 2016-09-22:
I ran on Linux with MKL and with OpenBLAS and I can't replicate, sorry :(

If you're stuck on it for a long time, I can try to dust off my old OSX laptop.

kingjr, 2016-09-22:
I can't reproduce it on linux either

dengemann, 2016-09-22:
can you set mkl threads to 1? it can play badly with joblib.

agramfort, 2016-09-23:
it's problem with accelerate framework.

use mkl to fix the pb or switch to py3k

Eric89GXL, 2017-02-14:
I'm going to close this for now. @teonbrooks if you can replicate with latest `master` versions of all these libraries we can take another look

closed 2017-02-14 by Eric89GXL