mne-tools/mne-python #3619: realtime (ftclient) example fails to read FieldTrip header

jjnurminen, 2016-09-27:
Hi,
I tried the example `ftclient_rt_compute_psd` with a TRIUX and `neuromag2ft`. It fails with
`ValueError: No channels match the selection.`

I traced it down to a problem reading the FieldTrip header from the raw data buffer. Here is a minimal example:

```
from mne.externals.FieldTrip import Client as FtClient
``
ftc = FtClient()
ftc.connect('alpha', 1972)
hdr = ftc.getHeader()
ftc.disconnect()
print hdr.labels
```

If I run `neuromag2ft` on alpha, start data acquisition and run the example, it will print an empty list, i.e. looks like it cannot read the channel info. Meanwhile in FieldTrip,  I can do `ft_read_header('buffer://alpha:1972')`
and I get the channel labels without any problems.

Curiously, if I stop the acquisition and start again, FtClient also starts to work and provides the correct channel info.

Any idea?

jasmainak, 2016-09-27:
can you share a small data file (even a couple of seconds would do) of recording from the same system?

jjnurminen, 2016-09-27:
I think I found a solution, namely a typo/bug in externals/FieldTrip.py.
I will submit a PR.

jjnurminen, 2016-09-27:
FieldTrip.py line 312 says:
`                if offset + chunk_len < bufsize:`
I'm thinking it should probably say:
`                if offset + chunk_len > bufsize:`

With this change, it starts to work.

jasmainak, 2016-09-27:
You will need to update it upstream with the FieldTrip folks if it's a bug in externals: https://github.com/fieldtrip/fieldtrip/blob/master/realtime/src/buffer/python/FieldTrip.py

jjnurminen, 2016-09-27:
Right, thanks @jasmainak. So I think we can close this one.

jjnurminen, 2016-09-27:
@jasmainak I discovered another bug. Will submit a PR.

jjnurminen, 2016-09-27:
(closing this anyway since title is not accurate anymore)

closed 2016-09-27 by jjnurminen