mne-tools/mne-python #3621: n_fft issue

ImmanuelSamuel, 2016-09-27:
Changing n_fft does not change the resolution of the power obtained when using
`mne.time_frequency.psd_welch`

Eric89GXL, 2016-09-27:
is `n_fft > n_samples` for your signal?

ImmanuelSamuel, 2016-09-27:
yes

Eric89GXL, 2016-09-27:
Read the docstring of `n_fft` more carefully, it says that if `n_fft > n_samples` , then `n_samples` will be used, i.e. `n_fft = min(n_fft, n_samples)`. Make sense?

ImmanuelSamuel, 2016-09-27:
Yes I see that sorry about that. I had previously had n_fft  > n_samples and I got the zero-padded result. Not qute sure what function I used. Does mne support zero-padding or should I have to zero-pad the data before turning it into an Epoch?

ImmanuelSamuel, 2016-09-27:
I know that tfr_* does that but what if I needed padding without the need to do time-frequency analysis? In this case where I fix the time bounds computation time is faster.

mmagnuski, 2016-09-28:
IIRC psd_welch does not support padding and I also remember myself expecting at the beginning that `n_fft > n_samples` does zero padding. 
The `scipy.signal.welch` function that `psd_welch` is using supports zero-padding however: `nperseg` argument controls segment length and `nfft` is used for zero-padding, when needed.

Eric89GXL, 2016-09-28:
I suppose we could change it so that if `n_fft > n_samples` it zero-pads, seems more consistent. Not sure how to do it in a nice way from a non-API-breaking standpoint, though. We might need to introduce a kwarg :(

mmagnuski, 2016-09-28:
In some cases one may still want a welch window that is shorter than n_samples and zero-padding too, so another kwarg seems necessary (`pad_to`, `pad`, `zero_pad` ?).

agramfort, 2016-09-28:
I agree

Eric89GXL, 2017-02-16:
@mmagnuski did you change this behavior in #3823? If so can you update `whats_new.rst` to say so, and we can close this?

mmagnuski, 2017-02-16:
No, I have a local branch for this. I will try to push it tomorrow.

Eric89GXL, 2017-02-16:
Great thanks for fixing these issues

PR #4003 (2017-02-24): [MRG+2] ENH: padding kwarg for psd_welch
Fixes #3621 
by adding `padding` kwarg to `psd_welch`.
I'll add an entry to whats_new once comments are resolved and CIs green.
closed 2017-02-24 by Eric89GXL