mne-tools/mne-python #3623: BUG: sklearn cannot find LeaveOneLabelOut

Eric89GXL, 2016-09-28:
On latest `sklearn` I get this error in `master`:
```
======================================================================
ERROR: Test time generalization decoding
----------------------------------------------------------------------
Traceback (most recent call last):
  File "/home/larsoner/custombuilds/nose/nose/case.py", line 197, in runTest
    self.test(*self.arg)
  File "/home/larsoner/custombuilds/mne-python/mne/utils.py", line 807, in dec
    return function(*args, **kwargs)
  File "/home/larsoner/custombuilds/mne-python/mne/decoding/tests/test_time_gen.py", line 61, in test_generalization_across_time
    from sklearn.model_selection import (KFold, StratifiedKFold,
ImportError: cannot import name LeaveOneLabelOut
```
@kingjr @agramfort were we using a deprecated feature, or is this a sklearn bug?

Eric89GXL, 2016-09-28:
Actually looks like that class doesn't even exist in `master`:

http://scikit-learn.org/dev/modules/classes.html

Eric89GXL, 2016-09-28:
Ahh, it's now called `LeaveOneGroupOut`

kingjr, 2016-09-28:
oh god, they call it "LeaveOneGroupOut" now :/ We urgently need to deprecate classes that handle the cv, it's causing too much bleeding edge.

agramfort, 2016-09-28:
ok but needs a backport :(

Eric89GXL, 2016-09-28:
Do we need it? We only have the problem on bleeding-edge sklearn. We'll probably get 0.14 out before they get out 0.19 or 0.20

agramfort, 2016-09-28:
we'll have a problem with 0.18 released imminently ...

Eric89GXL, 2016-09-28:
I'm on 0.19-dev, are you sure it's a problem with 0.18?

agramfort, 2016-09-28:
yes...

Eric89GXL, 2016-09-28:
It only shows up in `mne/decoding/tests/test_time_gen.py` (not core code), so maybe not worth it? :)

agramfort, 2016-09-28:
unless it makes debian packaging break....

Eric89GXL, 2016-09-28:
Argh, yes it could. :( I can do it after #3622 is in

Eric89GXL, 2016-09-29:
Indeed the problem is now in `master`:

https://travis-ci.org/mne-tools/mne-python/jobs/163710863

It is fixed by #3622, so I'll backport to 0.13 once that's in.

closed 2016-10-02 by Eric89GXL

agramfort, 2016-10-02:
can you backport?

Eric89GXL, 2016-10-03:
done

PR #3622 (2016-10-02): MRG: Deprecations
Doing my part for release :)

This is all the deprecations for 0.14.

Ready for review/merge from my end.

Closes #3623.