mne-tools/mne-python #3624: BUG: GAT shouldn't take non-partition cv

kingjr, 2016-09-28:
We should create an error when the cv is not a partition (e.g. ShuffleSplit with overlapping tests sets). We currently just iteratively replace the y_pred[test].

Eric89GXL, 2016-09-28:
Are these warnings related (from plot_decoding_csp_eeg.py)?
```
/home/larsoner/.local/lib/python2.7/site-packages/sklearn/lda.py:4: DeprecationWarning: lda.LDA has been moved to discriminant_analysis.LinearDiscriminantAnalysis in 0.17 and will be removed in 0.19
  "in 0.17 and will be removed in 0.19", DeprecationWarning)
/home/larsoner/.local/lib/python2.7/site-packages/sklearn/cross_validation.py:44: DeprecationWarning: This module was deprecated in version 0.18 in favor of the model_selection module into which all the refactored classes and functions are moved. Also note that the interface of the new CV iterators are different from that of this module. This module will be removed in 0.20.
  "This module will be removed in 0.20.", DeprecationWarning)
```

kingjr, 2016-09-28:
nope, different issue.

The second warning is linked to sklearn big change of cross_validation to model_selection module.

closed 2016-12-31 by agramfort