mne-tools/mne-python #3629: CUDA crashes with large arrays?

kambysese, 2016-09-30:
Hitting error on previously processed after updating to upstream https://github.com/mne-tools/mne-python/commit/3b09ed062226bbe017243071f450e0254fc3a769 data during filter/resampling operations. CUDA complains about array size after initializing:

> CUDA not used, could not instantiate memory (arrays may be too large), falling back to n_jobs=1

Then fails with:

> Traceback (most recent call last):
>   File "/home/ktavabi/Projects/shiva/shiva/process.py", line 163, in <module>
>     print_status=True
>   File "/home/ktavabi/Projects/mnefun/mnefun/_mnefun.py", line 591, in do_processing
>     outs[ii] = func(p, subjects, run_indices)
>   File "/home/ktavabi/Projects/mnefun/mnefun/_mnefun.py", line 1765, in do_preprocessing_combined
>     phase=p.phase, fir_window=p.fir_window, pick=True)
>   File "/home/ktavabi/Projects/mnefun/mnefun/_mnefun.py", line 1701, in _raw_LRFCP
>     r.resample(sfreq, n_jobs=n_jobs_resample, npad='auto')
>   File "<string>", line 2, in resample
>   File "/home/ktavabi/Projects/mne-python/mne/utils.py", line 708, in verbose
>     return function(*args, **kwargs)
>   File "/home/ktavabi/Projects/mne-python/mne/io/base.py", line 1286, in resample
>     window=window, n_jobs=n_jobs))
>   File "<string>", line 2, in resample
>   File "/home/ktavabi/Projects/mne-python/mne/utils.py", line 708, in verbose
>     return function(*args, **kwargs)
>   File "/home/ktavabi/Projects/mne-python/mne/filter.py", line 1804, in resample
>     cuda_dict)
>   File "/home/ktavabi/Projects/mne-python/mne/cuda.py", line 341, in fft_resample
>     x_fft *= W[np.arange(1, len(x) + 1) // 2].real
>   File "/home/ktavabi/anaconda2/lib/python2.7/site-packages/pycuda/gpuarray.py", line 874, in __getitem__
>     raise IndexError("invalid subindex in axis %d" % index_axis)
> IndexError: invalid subindex in axis 0

Eric89GXL, 2016-10-01:
So this doesn't happen on older MNE-Python (like 0.12) but does on `master`?

This shows that there is definitely a bug with our handling of the case where CUDA seems like it will work during init, but fails later -- but I'm curious whether the fallback worked before. I suspect you'll still hit the could-not-instantiate error on older MNE.

kambysese, 2016-10-03:
@Eric89GXL No, I've never seen this one before, and I am not sure where/how to go back to figure out how the "bug" was introduced. I am now on branch `maint/0.12` and I can reproduce the error with the mne sample data using `raw.resample(200, n_jobs='cuda', npad='auto')`; I also get the error when running `analysis.py` with `gen_ssp=True` on the mnefun funloc datasets.

kambysese, 2016-10-03:
Oh God... 

> pycuda.driver.CompileError: nvcc preprocessing of /tmp/tmpqs5L7d.cu failed
> [command: nvcc --preprocess -arch sm_30 -I/home/ktavabi/anaconda2/lib/python2.7/site-packages/pycuda/cuda /tmp/tmpqs5L7d.cu --compiler-options -P]
> [stderr:
> In file included from /usr/local/cuda-7.5/bin/../targets/x86_64-linux/include/cuda_runtime.h:76:0,
>                  from <command-line>:0:
> /usr/local/cuda-7.5/bin/../targets/x86_64-linux/include/host_config.h:115:2: error: #error -- unsupported GNU version! gcc versions later than 4.9 are not supported!
>  #error -- unsupported GNU version! gcc versions later than 4.9 are not supported!

This is probably yet another CUDA/pyCuda configuration issue on my end

kambysese, 2016-10-03:
@Eric89GXL It looks like the error is due to an incompatibility between CUDA and gcc version (>4.9) in Ubuntu 16.04. See [here](https://github.com/NervanaSystems/neon/issues/246) and [here](https://stackoverflow.com/questions/6622454/cuda-incompatible-with-my-gcc-version#comment56532695_8693381) for related conversations and workaround, and [here](https://youtu.be/_36yNWw_07g?t=10s) for general relief.

closed 2016-10-03 by kambysese