mne-tools/mne-python #3631: ENH: example with noise levels from empty room

Eric89GXL, 2016-10-03:
People often show noise levels of MEG systems as e.g. 10 fT/Hz**0.5. We should probably have an example where we do this with data. I'm thinking it would be cool to have it for empty-room data from different sites and/or systems, too, maybe eventually with e.g. sythetic gradiometry and/or compensation grades for systems that support those.

I thought about showing how different denoising algorithms work for some of these datasets, but that might make the example too dense. But collecting these files would at least make it more tenable to do do those types of analyses. Other ideas welcome.

For starters, @kingjr can you get a good empty-room recording from your setup and put it somewhere for me to get?

kingjr, 2016-10-03:
Sure. You only want kit or all systems ?

Eric89GXL, 2016-10-03:
Whatever systems you have

SherazKhan, 2016-10-06:
I can provide latest one from Martinos (The quietest room in the world)

Eric89GXL, 2016-10-06:
I was planning on using the one from the sample dataset for MGH, but if you have a "more illustrative" example for me to use from The Quietest Room in the World™ feel free to upload one and shoot me a link

SherazKhan, 2016-10-07:
May be I share THE  [world record MEG empty recording from Martinos](http://www.guinnessworldrecords.com/world-records/weakest-magnetic-field-measured)

Eric89GXL, 2016-10-11:
Oh, I thought that was a joke until I read the link. Awesome. Yes please :)

PR #4018 (2017-02-26): MRG: More PSD modes
Closes #3631.

This changes our normalization and units to be clearer in `raw.plot_psd`. In the case where `dB=False`, it plots amplitudes instead of power, which seems to be how people prefer to view these things (1/sqrt(Hz)).

I'll paste images of new data when they come up on CircleCI.

~~Not ready for MRG because this will break some `Epochs` plotting. I'm going to wait to fix that until people are happy with the raw plots in case I need to add other things along the way.~~
closed 2017-02-26 by agramfort