mne-tools/mne-python #3635: Issues with activation disappearing during source localization

misanyu, 2016-10-04:
We are attempting to run source localization on our MEG data set, via jupyter notebook.

We get as far as:

```
subjects_dir = data_path + '/subjects'
brain = stc.plot(surface='inflated', hemi='lh', subjects_dir=subjects_dir)
brain.set_data_time_index(350)
brain.scale_data_colormap(fmin=8, fmid=12, fmax=15, transparent=True)
brain.show_view('lateral')

brain.save_image('dspm.jpg')
from IPython.display import Image
Image(filename='dspm.jpg', width=600)
```

Every time we run this, the output window pops up, shows the activation on the brain, then the activation disappears and the window crashes.

We tried to save the image as well, and the activation is not present on the image (it's just the template brain).

Thank you for any help!

jasmainak, 2016-10-04:
It sounds like a backend issue. Did you try `%matplotlib qt` ?

jaeilepp, 2016-10-05:
It might be because of the windows being automatically closed at the end of a script. The backends behave differently in different modes of interactivity (I never really use notebooks so I'm not sure about this). You could try blocking the code somehow. For instance you could do ``raw.plot(block=True)`` at the end to see if it works.

misanyu, 2016-10-05:
Thank you for your suggestions!

We tried %matplotlib qt and it at least caused the window not to crash a few seconds after it pops up. However, the activation is still initially appearing then disappearing.

raw.plot (block=True) didn't work because raw is not defined, as we are not working with raw data.

jasmainak, 2016-10-05:
Can you tell us what you get when you type this:

```py
>>> mne.sys_info()
```

misanyu, 2016-10-05:
```
Platform:      Linux-4.4.0-38-generic-x86_64-with-debian-stretch-sid
Python:        2.7.12 |Anaconda custom (64-bit)| (default, Jul  2 2016, 17:42:40)  [GCC 4.4.7 20120313 (Red Hat 4.4.7-1)]
Executable:    /home/lyam/anaconda2/bin/python

mne:           0.12.0
numpy:         1.9.3 {lapack=mkl_lapack95_lp64, blas=mkl_intel_lp64}
scipy:         0.17.1
matplotlib:    1.4.3

sklearn:       0.16.1
nibabel:       2.1.0
nitime:        Not found
mayavi:        4.4.0
nose:          1.3.7
pandas:        0.17.1
pycuda:        Not found
skcuda:        Not found
```

jasmainak, 2016-10-05:
Did you try using `time_viewer=True` ?

Eric89GXL, 2016-11-18:
Doing just `Image(filename='dspm.jpg', width=600)` without assigning it (like `img = Image(filename='dspm.jpg', width=600)`) is probably causing it to be immediately garbage-collected and thus closed

closed 2016-11-18 by Eric89GXL