mne-tools/mne-python #3636: maxfilter error, extra space ambiguity between raw and ctc, channel names

SherazKhan, 2016-10-05:
I am trying to run maxfilter using mne-python on my MGH raw data, I found small discrepancy in the channel names, there is a missing space in the info['ch_names'] in the MGH data, whereas ctc file have a space between the channel number and the MEG. 

This missing space in the new MGH MEG data is after the electronics upgrade whereas ctc file was calculated with older electronics. 

Please download the cropped raw file from [cropped raw file](http://martinos.org/~sheraz/mgh_fif_file_after_electronics_upgrade_raw.fif)

Some code to reproduce the problem

```
import mne
from mne.preprocessing import maxwell_filter
data_path = mne.datasets.sample.data_path()
ctc_fname = data_path + '/SSS/ct_sparse_mgh.fif'
fine_cal_fname = data_path + '/SSS/sss_cal_mgh.dat'
fname_raw = 'mgh_fif_file_after_electronics_upgrade_raw.fif'
raw = mne.io.read_raw_fif(fname_raw,preload=True)
raw_sss = maxwell_filter(raw, cross_talk=ctc_fname, calibration=fine_cal_fname)
```

I think, fix is pretty simple by putting a check before line 324 of maxwell.py and doing something like this
`stripped_ctc_chs = ["".join(ctc_ch.split()) for ctc_ch in ctc_chs]`

However I think this missing space in raw or extra space in ctc, needs to be handle somewhere lower, may be at io.proc_history._read_ctc()  ?

Eric89GXL, 2016-10-05:
Sneaky bug. We have a filename normalization function `_clean_names` to deal with this common problem.

The easiest fix is probably to call the normalization function on both the CTC matrix and raw channel name list where they get compared in the `maxwell_filter` code. Do you have time to make a PR? The test can be simple enough as using the MGH cross-talk file in testing data (which should have spaces) with one of the test files from UW (which will not have spaces), e.g. `test_move_anon_raw.fif`.

SherazKhan, 2016-10-05:
Thanks Eric, since I am very indebted to mne-python, this is my time to pay dues, I will give it a try.

agramfort, 2016-10-05:
👍 👍 👍 👍 👍 👍 👍 👍 👍 👍 👍 👍 👍 👍 👍 👍 👍 👍 👍 👍 👍 👍 👍 👍 👍 👍 👍 

go go @SherazKhan !

SherazKhan, 2016-10-05:
Forgive me, this is my first try for PR for any public repository.

Eric89GXL, 2016-11-18:
This one has been fixed

closed 2016-11-18 by Eric89GXL