mne-tools/mne-python #3639: mixed_norm with volume source space?

jhouck, 2016-10-05:
I'd tried an initial pass at this using the sample data, following run_meg_volume_tutorial.py and then some slight modifications to plot_mixed_norm_inverse.py to use the volume forward and inverse.  In the first pass it didn't like the dSPM weights so to simplify things I set weights=None.  This ran through a few iterations, reported that the debiasing had converged, and then crashed with what looks like one of those errors that happens when you try to do apply a function intended for a surface to a volume (the error was with 'rh_vertno' in _make_sparse_stc).

Is this an error on my part or is mixed_norm (or I suppose the inverse_sparse functions in general) really not intended for use with volumes?

Eric89GXL, 2016-10-05:
I think nobody has tried it yet, but it should hopefully be simple enough
to fix. I can take a look next week.

jhouck, 2016-10-05:
Great, thanks! Good to know I hadn't attempted anything... unnatural.

agramfort, 2016-10-05:
@joewalter has done this but viz is currently the problem.

joewalter, 2016-10-05:
Hi, the fix is really easy, and I have code ready for this. However, my problem is that the visualization for sparse estimates is not ideal at the moment. Having the dipole plotting methods @Eric89GXL is working on would be awesome. The other issue with volume source spaces and mixed-norms is the depth weighting. I don't know if our simple gain matrix normalization is the best way to do it. But it is currently the only method available. Comparing depth weighting approaches and finding a probably better solution will require a bit of work.

Shall I open a pull request now? Even if the output of the examples is not optimal?

Eric89GXL, 2016-10-05:
> Shall I open a pull request now? Even if the output of the examples is not optimal?

+1 from my end, especially if you add a `.. note::` in the sparse solver(s) about the possibly insufficiency of the current depth-weighting scheme for non-cortical source spaces..

agramfort, 2016-10-05:
early WIP PR are always useful ... at least it exists

joewalter, 2016-10-05:
WIP PR sent

agramfort, 2017-07-22:
you can return dipoles with mixed_norm and use  plot_dipole_locations to overlay on MRI

closed 2017-07-22 by agramfort