mne-tools/mne-python #3642: can not zoom while importing data using mne.io.RawArray

abhishekupadhyaya, 2016-10-05:
I am not able to zoom in the plot when I create a raw instance using numpy array. However, when I directly load an edf file and create its instance, I can zoom in the plot.

Could you please clarify on this issue?

Thank you!

mmagnuski, 2016-10-05:
What plot are you reffering to? If `raw.plot()` - then you should clarify what you mean by zooming.

Eric89GXL, 2016-10-05:
Did you set the channel types properly in `create_info`? When plotting, we apply scaling based on channel type, so reading data directly as EDF / FIFF will generally work because channel types are set properly. But creating data with `RawArray` you need to be careful about what channel type you set in `create_info`. Something like `'misc'` probably won't be scaled at all, whereas `'eeg'` will be scaled by ~1e6

abhishekupadhyaya, 2016-10-06:
I have set the channel types properly. I am not sure where exactly the problem is but it might be something to do with mpl.use(‘TkAgg’) in the backend of my machine.

Thank you for your input

Eric89GXL, 2016-10-06:
> it might be something to do with mpl.use(‘TkAgg’) in the backend of my machine.

But it works with EDF data, and you are using the same backend for both use cases right?

Can you make a minimal example that reproduces the issue?

Eric89GXL, 2016-11-18:
Closing due to non-response, happy to reopen if it can be reproduced

closed 2016-11-18 by Eric89GXL