mne-tools/mne-python #3647: BUG: Averaging/plotting EpochArray/EvokedArray from ICA.get_sources leads to errors

kdoelling1919, 2016-10-06:
I'm using `ICA.get_sources()` to output an `EpochArray`: `epochs_ic`. If I try`evoked_ic=epochs_ic.average()` I get the following error:
```python
/Users/keithd/mne-python/mne/io/pick.pyc in pick_info(info, sel, copy)
    387         return info
    388     elif len(sel) == 0:
--> 389         raise ValueError('No channels match the selection.')
    390 
    391     info['chs'] = [info['chs'][k] for k in sel]

ValueError: No channels match the selection.
```
I can deal with this issue by explicitly giving the picks: something like `evoked_ic = epochs_ic.average(picks=np.arange(ncomps))`

Then if I try to do a butterfly plot of the evoked data I get a different error 
```python
/Users/keithd/anaconda2/lib/python2.7/site-packages/matplotlib/gridspec.pyc in __getitem__(self, key)
    152                     k1 += nrows
    153                 if k1 >= nrows or k1 < 0 :
--> 154                     raise IndexError("index out of range")
    155                 row1, row2 = k1, k1+1
    156 

IndexError: index out of range
```

That error I can't fix with the previous trick. Below is some code that reproduces both errors with sample data. I think both errors are likely related to the way ICA channels are picked in `/mne/io/picks.py`. It seems as though the ICA channel type is not listed at all as an option in `_PICK_TYPES_DATA_DICT`. I think I could fix this myself if others agree it is indeed a bug and I'm not missing something obvious. Let me know what you think!  

***Sample Code to reproduce error:***
```python
#import mne and sample data
import sys
import mne
from mne.datasets import sample  # noqa
data_path = sample.data_path()
raw_fname = data_path + '/MEG/sample/sample_audvis_filt-0-40_raw.fif'
raw = mne.io.read_raw_fif(raw_fname, add_eeg_ref=False)

# get the events
events = mne.find_events(raw, stim_channel='STI 014')
event_id = dict(aud_l=1, aud_r=2)  # event trigger and conditions
tmin = -0.5  # start of each epoch (500ms before the trigger)
tmax = 0.5  # end of each epoch (500ms after the trigger)
# pick just the meg channels
picks = mne.pick_types(raw.info, meg=True, eeg=False, eog=False, stim=False,
                       exclude='bads')
# make epochs
epochs = mne.Epochs(raw, events, event_id, tmin, tmax, proj=True, picks=picks,
                    preload=True,add_eeg_ref=False)
# run ICA on the epochs
ncomps = 24
ica=mne.preprocessing.ica.ICA(max_pca_components=ncomps)
ica.fit(epochs)
# put sources into epoch array
src_data=ica.get_sources(epochs)
# try to plot evoked array
evoked_ic = src_data.average(); # get error here, can be avoided by including picks=np.arange(ncomps)
evoked_ic.plot() #then I can get an error here. This is not fixable by the same addition
```

jaeilepp, 2016-10-07:
Fix in https://github.com/mne-tools/mne-python/pull/3649
This happened because the ICA components are assigned as misc channels. Previously misc channels could not be plotted in evoked.

kdoelling1919, 2016-10-07:
Thanks @jaeilepp !! This fixes the error with evoked_ic.plot() 

It doesn't fix however the error generated from `evoked_ic = src_data.average()`

While changing it to `evoked_ic=src_data.average(picks=np.arange(ncomps))` gets around the error, it seems like we should fix that bug as well in the first place. I guess, it should be a pretty similar issue.

jaeilepp, 2016-10-10:
The documentation of [average](http://mne-tools.github.io/dev/generated/mne.Epochs.html?highlight=average#mne.Epochs.average), says that only data channels are picked by default. User can also give picks to include ``misc`` channels. I'm not sure if we want to change that.

closed 2016-10-10 by jaeilepp

jona-sassenhagen, 2016-10-24:
Maybe we should amend the warning on epoching if the object is detected to be ICA sources. I.e., the warning could tell the user about @kdoelling1919 's trick or something like that.

kdoelling1919, 2016-10-25:
I'm happy to do this, if others agree it would be useful.

agramfort, 2016-10-26:
+1 to improve docstrings