mne-tools/mne-python #3657: 'ascii' codec can't encode  while in test_add_slider_to_section  on Debian stable

yarikoptic, 2016-10-11:
Haven't looked in detail but that is on Debian jessie (stable) with python 2.7.9

```
======================================================================
ERROR: Test adding a slider with a series of images to mne report.
----------------------------------------------------------------------
Traceback (most recent call last):
  File "/usr/lib/python2.7/dist-packages/nose/case.py", line 197, in runTest
    self.test(*self.arg)
  File "/build/python-mne-0.13+git5-g028f2ee/mne/tests/test_report.py", line 247, in test_add_slider_to_section
    report.save(op.join(tempdir, 'report.html'), open_browser=False)
  File "/build/python-mne-0.13+git5-g028f2ee/mne/report.py", line 1330, in save
    self._render_toc()
  File "<string>", line 2, in _render_toc
  File "/build/python-mne-0.13+git5-g028f2ee/mne/utils.py", line 708, in verbose
    return function(*args, **kwargs)
  File "/build/python-mne-0.13+git5-g028f2ee/mne/report.py", line 1437, in _render_toc
    sectionvars=self._sectionvars)
  File "/build/python-mne-0.13+git5-g028f2ee/mne/externals/tempita/__init__.py", line 178, in substitute
    result, defs, inherit = self._interpret(ns)
  File "/build/python-mne-0.13+git5-g028f2ee/mne/externals/tempita/__init__.py", line 189, in _interpret
    self._interpret_codes(self._parsed, ns, out=parts, defs=defs)
  File "/build/python-mne-0.13+git5-g028f2ee/mne/externals/tempita/__init__.py", line 217, in _interpret_codes
    self._interpret_code(item, ns, out, defs)
  File "/build/python-mne-0.13+git5-g028f2ee/mne/externals/tempita/__init__.py", line 241, in _interpret_code
    out.append(self._repr(base, pos))
  File "/build/python-mne-0.13+git5-g028f2ee/mne/externals/tempita/__init__.py", line 349, in _repr
    raise(e)
UnicodeEncodeError: 'ascii' codec can't encode character u'\xa0' in position 14245: ordinal not in range(128)

```

jaeilepp, 2016-10-12:
I'm having trouble reproducing this. Is there any more information about the platform?

yarikoptic, 2016-10-14:
Let me know which detail you would like to have.  It is a Debian jessie (amd64 in this case), clean (no other than necessary dependencies installed).  Just replicated it in my env interactively and can answer any directed question. the content which gets encoded is too big to paste here... might as well be just a bug in python itself but probably unlikely.  I didn't try but probably could be replicated in a debian docker/ the easiest is probably to use neurodebian docker image for jessie, smth like 

```
sed -e 's,deb ,deb-src ,g' /etc/apt/sources.list.d/neurodebian.sources.list > /etc/apt/sources.list.d/neurodebian.sources-src.list
apt-get -qq update ; apt-get -y build-dep python-mne
apt-get install -y git
git clone ...
```
didn't try yet in full ;)

yarikoptic, 2016-11-11:
ha -- an now appears on debian sid

agramfort, 2016-11-12:
can someone on a debian or at least a linux system replicate ?

closed 2016-11-18 by Eric89GXL