mne-tools/mne-python #3663: BUG: extra 0s when Epochs(tmin=0)

kingjr, 2016-10-14:
giving a tmin = 0 pads 0. at the beggining of the epochs:

```
import mne
import numpy as np
fname = mne.datasets.sample.data_path() + '/MEG/sample/sample_audvis_filt-0-40_raw.fif'
raw = mne.io.read_raw_fif(fname)
event = np.array([[10000, 0, 1], [10001, 0, 1]], int)
epochs = mne.Epochs(raw, event, tmin=0.)
print(epochs.get_data()[:, :, 0])
```

which can be uncool in some use cases (baselining, re-sticking the data as we do in xdawn, etc)

agramfort, 2016-10-15:
to be it does not pad it just baselines using one sample so you see zeros.
right?

mmagnuski, 2016-10-15:
Yes, I've seen it when writing `_segment_raw` - it's a single-sample baseline :)

kingjr, 2016-10-18:
> to be it does not pad it just baselines using one sample so you see zeros.
right?

right, stupid me

closed 2016-10-18 by kingjr