mne-tools/mne-python #3664: Q? Epochs default SSP

kingjr, 2016-10-14:
What is the motivation of having proj=True by default when creating epochs?

It's been a difficult to track annoyance in multiple cases for me :(. IMO this kind of operation should be explicit asked by the user.

Eric89GXL, 2016-10-14:
I agree it would be good, but I'm not sure if the deprecation cycle is
worth it

jona-sassenhagen, 2016-10-14:
@kingjr you essentially want `proj=` to default to something so it raises if it's not user set?

kingjr, 2016-10-14:
@jona-sassenhagen no I want proj=False by default. I keep forgetting that when I just epoch my raw data and extract the numpy array for subsequent analyses, an SSP has actually been performed (which for instance screw me up for an hour while trying to figure re-implement @Eric89GXL's OTP). I've encountered this before with Riemannian's functions, because the covariance isn't positive definite anymore.

I think I'm just going to write my own Epochs class from no on; it's just crazy to have to check the baseline and the proj every time.

jona-sassenhagen, 2016-10-14:
Ah makes sense. I'm not sure proj=True is the least-surprise case for EEG either.

agramfort, 2016-10-15:
sorry it was an early decision.

now for you @kingjr why not removing the proj from info on read if you
don't need them?

kingjr, 2016-10-18:
For now I'm fine, just explicitly set the Epochs params. It's more like an API problem IMO, but I understand how the inertia limits this kind of correction.

dengemann, 2016-10-20:
It was not so obvious that ssp projs would become obsolete for many. My
approach would also be: just delete them from the info if you have them in
your data. Working with neuromag data means handling SSP explicitly very
often. Maybe we should document this. Of course we have a bunch of such
parameters, e.g. baseline, which is rarely useful for time frequency/
resting state etc. and can give surprising results. I would baseline prefer
to be None by default. But inertia ...

Eric89GXL, 2017-02-14:
@kingjr also doing `raw.del_proj()` is a one-liner that should get you where you want to be. 

We'll keep this `proj=False` idea in mind when dealing with #2310. In the meantime I'll close this.

closed 2017-02-14 by Eric89GXL