mne-tools/mne-python #3665: ENH: decoding robust option in Scaler

kingjr, 2016-10-15:
Currently the `mne.decoding.Scaler` takes the mean and std across channels to normalize the data.

We should have an option to either accept and sklearn.transformer, a dict a string, so that we can get 

* default values for meeg, eeg etc e.g: `Scaler(info, method='default')` would apply `mne.cov._apply_scalings_user()`
* data-estimated standard scaling: i.e. current behavior i.e. `X/std-mean` over time
* data-estimated robust scaling : e.g. `Scaler(method=sklearn.preprocessing.RobustScaler())`

@kaichogami that should be an easy one if you're interested

kingjr, 2016-10-15:
Here s a function for the first case ('default'):

```

def scale(X, info, scalings=None):
    scalings = dict() if scalings is None else scalings
    from mne.io.pick import pick_info, _pick_data_channels, _picks_by_type
    from mne.cov import _check_scalings_user, _apply_scaling_array
    picks = _pick_data_channels(info)
    apply_picks_list = _picks_by_type(pick_info(info, picks))
    scalings = _check_scalings_user(scalings)
    n_epochs, n_channels, n_times = X.shape
    X = X.transpose(1, 0, 2)
    _apply_scaling_array(X, apply_picks_list, scalings)
    X = X.reshape(n_channels, n_epochs, n_times).transpose(1, 0, 2)
    return X
```

agramfort, 2016-10-15:
in terms of API I would prefer:

Scaler(method='robust')

I don't like composition like :

Scaler(method=sklearn.preprocessing.RobustScaler())

it makes API too complex. Let's avoid it if possible

kaichogami, 2016-10-15:
Sure, I will work on this.

kingjr, 2016-10-15:
>  Scaler(method='robust')
+10

kingjr, 2016-10-15:
Here is a gist
```

class Scaler(TransformerMixin, BaseEstimator):
    def __init__(self, info, scaler=None):
        from sklearn.preprocessing import RobustScaler
        from mne.io.pick import pick_info, _pick_data_channels, _picks_by_type
        self.scaler = RobustScaler() if scaler is None else scaler
        self.info = info
        picks = _pick_data_channels(self.info)
        self._pick_types = _picks_by_type(pick_info(self.info, picks))

    def fit(self, X, y=None):
        from sklearn.base import clone
        self.scalers_ = list()
        n_epochs, n_channels, n_times = X.shape
        X = X.transpose(1, 0, 2).reshape(n_channels, -1)
        for (_, picks) in self._pick_types:
            scaler = clone(self.scaler)
            scaler.fit(X[picks].T)
            self.scalers_.append(scaler)
        return self

    def transform(self, X):
        n_epochs, n_channels, n_times = X.shape
        X = X.transpose(1, 0, 2).reshape(n_channels, -1)
        for scaler, (_, picks) in zip(self.scalers_, self._pick_types):
            X[picks] = scaler.transform(X[picks].T).T
        X = X.reshape(n_channels, n_epochs, n_times).transpose(1, 0, 2)
        return X

    def fit_transform(self, X, y=None):
        return self.fit(X, y).transform(X)
```

Eric89GXL, 2017-02-22:
From #4007 it's clear that these operations are done on a per-time-point (within each epoch, too) basis. This isn't really what's expected.

Eric89GXL, 2017-02-22:
The API should probably be `scalings='empirical' | 'standard'` or something, since elsewhere `scalings=None` means "use the default dict".

Eric89GXL, 2017-02-27:
I suspect we should fix this before release.

Given that `Scaler` never really worked as anyone expected, my vote is to fix it and mention it in API instead of making another class.

@agramfort @kingjr @jasmainak okay with you?

I can take over #3672 if necessary

agramfort, 2017-02-27:
ok

jasmainak, 2017-02-27:
@Eric89GXL +1, let us fix it

Eric89GXL, 2017-02-27:
Who is the "us" in that sentence? I know @kingjr is swamped, so I assume you're taking it over, then? It would be good to have it in ASAP so we can release if possible, this is one of only a few remaining blockers

jasmainak, 2017-02-28:
oops, no I didn't volunteer. Unless it can wait till the sprint :)

Eric89GXL, 2017-02-28:
Thanks for clarifying, I'll see if it's straightforward tomorrow hopefully

PR #4025 (2017-03-12): MRG: Fix Scaler
Closes #3665.

Ready for review/merge from my end. @kingjr can you verify this is the functionality you were thinking of? The tests should make it pretty clear.
closed 2017-03-12 by agramfort