mne-tools/mne-python #3666: ENH: MNE sample dataset 2.0, whishlist

SherazKhan, 2016-10-16:
At Martinos, we are collecting MNE sample dataset 2.0, I think it is right time to make wishlist, following are my suggestion, please feel free to add/edit. 

The plan is to release data publicly possibly through Amazon S3 and write an open access paper describing data along with example scripts on mne-python to showcase pre/post processing. 

**A.**   Simultaneous MEG and EEG (70 Channel, Easy Cap) on two subjects under following protocols, 
1. Median Nerve (Electrical stimulation).
2. Auditory Steady State (43Hz and 113Hz).
3. [Faces and Houses](http://www.pnas.org/content/110/8/3107.abstract).
4. Two sets of six minute resting state, eyes open fixating on red cross (start of session and other the at end).

We plan to repeat this simultaneous MEG and EEG possibly with EGI 256, low profile Gel Cap. This will serve two purposes;
1. Reproducibility of MEG Connectivity findings.
2. Possible advantages of simultaneous measurement of super high density EEG with MEG.

Measurement settings: Supine, 1 KHz Sampling, 0.03Hz to 330Hz Bandwidth, Horizontal and Vertical EOGs, ECG.

**B.** On The MRI side we plan to collect for each subject;
1. T1-weighted MPRAGE (3T).
2. Multi-Echo FLASH at both 5° and 30° flip angles (3T).
3. Repeat MEG/EEG tasks 2, 3 and 4 in fMRI (3T).
4. T1-weighted MPRAGE (_7T_), to explore 7T advantages in segmenting, orienting and localization of deep structures.
5. Diffusion-weighted MR using MGH connectome scanner at multiple B-values, to explore advantages of fusing DTI with MEG/EEG connectivity.

agramfort, 2016-10-17:
sounds like great plan.

any data is already acquired?

jona-sassenhagen, 2016-10-17:
As a language person, I'd say a few minutes passively listening to some language would be cool to have.

The HCP MEG includes Aesop's Fables.

SherazKhan, 2016-10-17:
@agramfort Actually one subject is already acquired, in MEG/EEG and structural MRI. 

Do you think 256 channel EEG will add something significant in source localization ?

SherazKhan, 2016-10-17:
@jona-sassenhagen We were thinking to have only those protocols that produce very strong evoked responses, and have been done to the depth by the MEG community. This will help in calibrating new signal processing routines that will be developed using  MNE-Python. 
Having said that this can be added, do you have a protocol that we can test ?

agramfort, 2016-10-17:
>
> @agramfort <https://github.com/agramfort> Actually one subject is already
> acquired, in MEG/EEG and structural MRI.
>

ok. let us know about your results. You can share an mne-report maybe or a
notebook

> Do you think 256 channel EEG will add something significant in source
> localization ?
>
that's the question I curious about too. I don't want to influence you by
giving my predictions :)

SherazKhan, 2016-10-17:
@agramfort putting 256 channel gel cap is real pain, it took us 90 mins, to have impedances below 5K in 95% of the channels, real pain :)

Since 256 channel EEG simultaneously with MEG has never been done before, so we thought this will be interesting :)

The subject MEG responses were textbook. I did not performed source localization with EEG, but I think you guys should give it a try for some enhancement in source localization. May be some simulation is needed for some theoretical backgrounding.

agramfort, 2016-10-17:
ok tell us how we can help

Eric89GXL, 2016-10-17:
That dataset looks great to me

jona-sassenhagen, 2016-10-17:
Entrainment to the speech envelope can produce very strong responses, see e.g. Lalor & Foxe 2010 EJN. But it's not evoked - you need to do crosscorrelation or otherwise calculate the impulse response function. We have tools in MNE-Python, but it would probably need manual coding elsewhere.

For the protocol, you basically need a few minutes of normal speech in any language the subjects can understand, ask people to listen to it, and that's it. As I mentioned, the HCP MEG stimuli might be appropriate - that would mean you could compare it to the 90+ HCP datasets.

johnsam7, 2016-10-17:
@agramfort @Eric89GXL How was your experience with Amazon S3 ?

SherazKhan, 2016-10-17:
@agramfort @Eric89GXL How was your experience with Amazon S3 ?

agramfort, 2016-10-17:
seems to work great for us

dengemann, 2016-10-17:
Yes no trouble. If you have questions I am currently hosting the mne data
on s3. and Sheraz, this sounds fantastic!! Keep me posted on this. I'd be
too very curious about the effect of high density EEG with MEG.

kingjr, 2016-10-17:
That would be a great addition.

Overall, I'm also +1 for having more cognitive tasks (language, working memory) and I would recommend to have a task that create induced oscillations (not steady state).

The tasks you propose are all focusing on early evoked responses, which can bias the preprocessing methods (e.g. apply high pass filtering instead of baseline correction).

A motor localizer could also be easy and helpful.

HTH

jona-sassenhagen, 2016-10-17:
>The tasks you propose are all focusing on early evoked responses, which can bias the preprocessing methods (e.g. apply high pass filtering instead of baseline correction).

Adding any forced-choice response tasks gives you a P3 for free, could use that.

kingjr, 2016-10-17:
As long as you avoid a P300 vs response confound I'm +1

jona-sassenhagen, 2016-10-17:
You can just time lock to response execution and you should catch the peak of the P3.

kingjr, 2016-10-17:
mmmh; this is kind of indirect control.. Better to orthogonolize response buttons by changing hand on every block

mmagnuski, 2016-10-17:
you get nice P300 even if motor response is delayed :)

SherazKhan, 2016-10-17:
@jona-sassenhagen @kingjr If you guys have some working cognitive tasks (language, working memory)  PsychoPy or Psychophysics Toolbox example script, it will be great. 

I am also incline to add [Pascal Fries Grating Gamma](http://jn.physiology.org/content/114/1/244), if there is enough interest in it.

kingjr, 2016-10-17:
@SherazKhan +1 for grating presented for ~ 400 ms, with a variable ITI, they induce nice low gamma responses, and can serve as a nice example of high freq responses in th MEG.

For the cognitive protocol, I'd be inclined to just re-use the HCP protocols; they're not ideal, but it would allow us to compare the results across centers/MEG devices.

mmagnuski, 2016-10-17:
+1 for grating

SherazKhan, 2016-10-17:
@dengemann Thanks, If you really want to try some fancy source localization, I can make the data available to you. Currently see the example script at [eeg_mff](https://github.com/SherazKhan/pyimpress/blob/master/examples/eeg_mff.py)

I will also start this as mne-python example.

SherazKhan, 2016-10-17:
@dengemann Can I ask how much it roughly cost your grant to have the data at Amzon S3

kingjr, 2016-10-17:
@SherazKhan you can apply to an amazon grant https://aws.amazon.com/grants/

Personally, I pay ~ 150 dollars of S3 storage / month for <10 MEG studies.

SherazKhan, 2016-10-17:
@kingjr Can you please suggest any paper that showcase ~ 400 ms, with a variable ITI, The one I am using currently has Grating on for 1000ms, with a jitter of 250 ms uniformly distributed. I am only doing two angles 30 and -30.

kingjr, 2016-10-17:
No paper in mind; just everyday practice. Low gamma is strongest for the first 500 ms, then starts to go down; Note that gamma responses depend on the spatial frequency:
http://cercor.oxfordjournals.org/content/25/9/2951.short

If you make gratings, might as well use more orientations. In my experience, cardinals generally elicit stronger responses, especially in EEG. These are the tuning curve of our latest study, with a masked target (17 ms presentation) and a probe (50 ms presentation).

[image]

[image]

It's pretty easy to isolate the orientation of the stim for each of the 6 possible orientations, and the response is sustained if it is task relevant.

SherazKhan, 2016-10-17:
@kingjr " for <10 MEG studies." Can I ask, How much it is in GBs ?

dengemann, 2016-10-17:
@SherazKhan <https://github.com/SherazKhan> storage itself is very cheap.
Costs mainly depend on how often you download the data via internet and
across regions (amazon has this concept of regions). If you need more
details let's discuss via phone(skype/hangout) also with @kingjr. We have
both used amazon extensively during the last year. Anyways calling would be
cool, since we did could not meet at Biomag.

dengemann, 2016-10-17:
... @SherazKhan <https://github.com/SherazKhan> forgot to emphasise it but
S3 storage costs are really negligible. Also checkout
http://calculator.s3.amazonaws.com/index.html . Computation
time/performance and mountable volumes are way more expensive.

SherazKhan, 2016-10-17:
@dengemann Thanks, we need your wisdow :) 

Amazon IO model is complicated, Its hard to predict IO, specially initially. I need to budget, lets talk, I am sherazpad1 on skype.

kingjr, 2016-10-17:
good idea; I'm available next week: jeanremi.king on skype or google hangout

SherazKhan, 2016-10-18:
@agramfort @dengemann @Eric89GXL @kingjr @mmagnuski @jona-sassenhagen;  I am also thinking to collect same data on each subject at MIT MEG, this might help us in cross validating decoding algorithms in a **site independent** way. 

I can easily do it, if there is enough interest ?, please let me know.

dengemann, 2016-10-18:
Yes +1

Eric89GXL, 2016-10-18:
+1

Then we need to collect from a CTF site to make it really interesting :)

SherazKhan, 2016-10-18:
@Eric89GXL Unfortunately closest CTF site is in NY.

MIT is just across the river :)

dengemann, 2016-10-18:
> MIT is just across the river :)

:)

Eric89GXL, 2017-02-20:
Closing this as a non-code-issue, but feel free to continue discussion as necessary

closed 2017-02-20 by Eric89GXL