mne-tools/mne-python #3669: Updating the MNE workflow image?

choldgraf, 2016-10-17:
I've found this image to be useful in explaining the general objects etc that exist in MNE:

[image]

I wonder if something like this exists for the kinds of analyses one can create in MNE as well? E.g., one for AverageTFR or EpochsTFR, one for time decoding, or maybe one for a function I don't even know about. Does anybody else think it'd be useful to either update the above diagram, or create complementary diagrams for modules etc?

kingjr, 2016-10-17:
raw => decoding

would be my ideal pipeline ;)

choldgraf, 2016-10-17:
That could be a big part of it :) I'm just thinking of how much people like that sklearn "what kind of analysis should I do" graph...this could be something like it

agramfort, 2016-10-18:
this graph is generated by :

https://github.com/mne-tools/mne-python/blob/master/doc/sphinxext/flow_diagram.py

improvements and suggestions welcome

Eric89GXL, 2016-10-18:
The coolest thing we could do is make some interactive webpage with a
dynamic graph with collapsible entries. Maybe D3 has something we could
use. Then we could probably put everything in one chart. Otherwise if we
try to put decoding, encoding (eventually), time frequency, ERP,
statistics, connectivity, etc. all in one static graph, it will probably be
too overwhelming. But we could try expanding the existing one a bit to see
if that is indeed the case.

Eric89GXL, 2016-10-18:
We could actually switch to a nice client-side rendered flowchart, which would be nice:

http://marvl.infotech.monash.edu/webcola/examples/gridifiedSmallGroups.html
http://flowchart.js.org/
http://live.yworks.com/yfiles-for-html/1.3/demos/databinding/demo.yfiles.binding.interactivegraphsource/index.html
http://live.yworks.com/demobrowser/index.html#Incremental-Hierarchic-Layout

Nice for a Sunday hack or documentation session

Eric89GXL, 2016-10-18:
Or Dagre-D3:

http://cpettitt.github.io/project/dagre-d3/latest/demo/interactive-demo.html
http://cpettitt.github.io/project/dagre-d3/latest/demo/clusters.html

choldgraf, 2016-10-18:
I wonder if the D3 example can be made interactive? If so that'd be pretty cool.

Nelle and I were thinking of organizing a little documentation hackathon / code sprint sometime in the coming months (not necessarily package specific). That could be a fun project.

kingjr, 2016-10-21:
In this spirit, it would be great to make a workflow for the coregistration: when you don't have an elekta, it is pretty difficult to get around the headshape/marker/hpi/fiducial files correctly, what they do and which coordinate system they're expected to be.

choldgraf, 2016-10-25:
Is there a "todo" list for the documentation somewhere?

Eric89GXL, 2016-10-25:
#1495

larsoner, 2023-10-01:
Closing in favor of #6778

closed 2023-10-01 by larsoner