mne-tools/mne-python #3678: ENH: HCP-MMP1.0 parcellation?

Eric89GXL, 2016-10-21:
There is a nice new parcellation of cortex available:

https://doi.org/10.1038/nature18933
https://balsa.wustl.edu/WN56

And a kind soul has already converted it to fsaverage:

https://figshare.com/articles/HCP-MMP1_0_projected_on_fsaverage/3498446/2

It would be nice to be able to do:
```
labels = read_labels_from_aparc('fsaverage', 'HCP-MMP1.0')
```
But that won't work without putting the files in `$SUBJECTS_DIR/fsaverage/label`. So we could do something like:
```
mne.datasets.hcp_mmp.install()
```
That will download the files to `$MNE_DATA/HCP-MMP1.0` and then copy the `*aparc*` files to `$SUBJECTS_DIR/fsaverage/label`. The workflow would then just be:
```
mne.datasets.hcp_mmp.install()
labels = read_labels_from_aparc('fsaverage', 'HCP-MMP1.0')
```
Sound good? Better ideas?

jona-sassenhagen, 2016-10-24:
I'd like having these labels available. API looks very simple. Good idea.

agramfort, 2016-10-25:
I would say :

mne.datasets.fetch_hcp_mmp_parcellation()

but otherwise ok for me.

Eric89GXL, 2016-10-25:
so `fetch_hcp_mmp_parcellation` would both download and install it (by default) to `$SUBJECTS_DIR/fsaverage/label`?

agramfort, 2016-10-25:
yes

PR #3698 (2016-10-26): MRG: Add HCPMMP1.0 parcellation
Closes #3678.
closed 2016-10-26 by agramfort