mne-tools/mne-python #3679: BUG: coregistration doesn't read fiducials anymore

kingjr, 2016-10-21:
Something has changed in mne.gui.coregistration. Now I can't load the fiducials file, it throws the following error:

[image]

I checked that this works fine with a previous version version

@christianbrodbeck

christianbrodbeck, 2016-10-21:
Can you read the file with `read_fiducials()`?

kingjr, 2016-10-21:
no:
```
----> 1 read_fiducials('/media/jrking/USB30FD/sub01/sub01_YFAALKWR_JA/_tmp_epoch_coreg.fif')

/home/jrking/mne-python/mne/io/meas_info.py in read_fiducials(fname)
    301         err = ("No coordinate frame was found in the file %r, it is probably "
    302                "not a valid fiducials file." % fname)
--> 303         raise ValueError(err)
    304 
    305     # coord_frame is not stored in the tag

ValueError: No coordinate frame was found
```

I had to revert back to ~ january 2016 to make it work.

Eric89GXL, 2016-10-21:
Want to try [git bisect] (http://webchick.net/node/99) to find the offending commit?

kingjr, 2016-10-21:
here is the file that worked with a previous version of mne https://ufile.io/8cc9

christianbrodbeck, 2016-10-21:
Well I guess we could silence that error? Warn instead? Which application saves fiducials without coordinate frame? Did we in the past?

kingjr, 2016-10-22:
The coordinate system was brainsight, usually used for TMS. I don't know much more, I'm not very skilled regarding coregistration functions and i/o.

Eric89GXL, 2017-02-14:
@kingjr I just tried to download your file but it has expired. Can you re-upload?

Eric89GXL, 2017-02-28:
@kingjr can you track down the offending file?

kingjr, 2017-03-07:
I'm closing this for now, as the last PRs fixed some problem. I'll re-open it if I encounter this issue again.

closed 2017-03-07 by kingjr