mne-tools/mne-python #3681: Serious issues with 'Linear Model'

kingjr, 2016-10-22:
I'm only reading this http://martinos.org/mne/stable/auto_examples/decoding/plot_linear_model_patterns.html

I have several serious concerns:

minor:
* we suggest to vectorize the data to apply a single linear model; (doesn't seem very smart, but ok)
* the code has many syntax issues, copy/paste from sklearn etc

major:
* we plot the patterns without rescaling the coefficients.
* there is no cross-validation the `_set_cv` is never called?
* there is no unit test

@agramfort @trachelr

trachelr, 2016-10-24:
Hi @kingjr 
the scale of the coefficients depends on the scale of the data. normalisation is applied before fitting the model... I dont see any issue here.
cross validation is available by model=LogisticRegressionCV in the LinearModel init function.
@agramfort unit test is needed for examples?

kingjr, 2016-10-24:
Regarding the CV, IIUC LogisticRegressionCV is a cross val to optimize the parameter (C) not a CV to estimate the effect size. Although the example doesn't explicitly score the data, the filters and patterns can only make sense if there's an effect of condition - a requirement which should be addressed via a cross_val_score IMO.

Regarding the unit tests, I was referring to the object LinearModel and the X.X^T function.

Regarding the scaling, I would think that if you want to source localize the pattern, you'd need to unscale the data, else the grads and mags (and the eeg if you fit a single model) won't be meaningfully related to one another anymore. @agramfort what's your view on this?

agramfort, 2016-10-25:
I don't see the pb. If you scale the features then you need to undo the
scaling to get patterns in signal space with good units.

agramfort, 2016-10-25:
it's the first iteration on this code/method so I am sure it needs some
love....

trachelr, 2016-10-25:
yesss it was quickly/kindly coded during a mne sprint last year and not heavily tested... so more contributions are welcome!
@kingjr I can write some unit tests to finish the job but not more since i'm not doing any decoding stuff anymore.

kingjr, 2016-10-25:
ok; let me first have a chat with Alex, I'm encountering weird things with the Haufe's trick; I wanna make sure I'm missing something before fixing the details.

PR #3949 (2017-02-21): MRG+1: get patterns coefs with example
After a couple of iterations, I believe the best way to deal with the extraction and plotting of spatial filters and spatial patterns is to i) ensure that we use the LinearModel class at the end of our pipeline and ii) use a function that can extract and inverse transform these coefficients.

This API allows a full sklearn compatibility, and avoid adding additional complex attributes to `SearchLight` and `GeneralizationLight` (filters_, inverse_filters, inverse_patterns etc)

```python
clf = make_pipeline(Vectorizer(), StandardScaler(), LinearModel(Logist [... cut]
closed 2016-10-25 by kingjr