mne-tools/mne-python #3697: BUG: _SearchLight when 'scoring=roc_auc' & y not in (0, 1)

kingjr, 2016-10-25:
@kaichogami we missed a necessary feature when adding the `scoring` parameter to the SearchLight in the last PR:  currently, we encounter an error if `scoring='roc_auc'` but `y` is not (0, 1):

```python
import numpy as np
from mne.decoding.search_light import _SearchLight
from sklearn.linear_model import LogisticRegression

sl = _SearchLight(LogisticRegression(), scoring='roc_auc')
X = np.random.rand(100, 20, 2)
y = np.random.randint(0, 2, 100)
sl.fit(X, y)
sl.score(X, y)  # works

y = np.random.randint(2, 4, 100)
sl.fit(X, y)
sl.score(X, y)  # crashes
```

I suggest to add a sklearn.preprocessing.LabelEncoder() at fitting and use it before scoring. @agramfort do you think it's the right approach?

kaichogami, 2016-10-27:
Ok, I'll work on that.

kaichogami, 2016-10-29:
I think we should use `LabelEncoder` only when number of classes of y is 2, in fit.

kaichogami, 2016-10-31:
PR in #3672 3717

kingjr, 2016-12-21:
I believe that this has to be solved at sklearn level: https://github.com/scikit-learn/scikit-learn/pull/6874

agramfort, 2016-12-21:
it's too complicated at sklearn level. We need to raise an error in MNE if
y is not [0, 1] and scoring == 'roc_auc' explaining that make_scorer with
the good pos_label set needs to be used

kingjr, 2016-12-21:
@agramfort that's bleeding edge IMO. It's not complicated to sort this at sklearn level, we just need to add LabelBinarizer within the roc if and only if len(set(y)) == 2

agramfort, 2016-12-22:
maybe but really the easy fix is in mne. We can do all this in our code.

PR #3874 (2016-12-22): MRG: roc_auc when y not in [0, 1]
Fixes error when scoring='roc_auc' and y is not in [0, 1]:

Should be ultimately solved in scikit-learn/scikit-learn#6874

Replaces #3717, Closes #3697
closed 2016-12-22 by agramfort