mne-tools/mne-python #3700: Add button to project electrodes onto scalp in  mne.gui.coregistration

noreun, 2016-10-26:
Since `mne.make_forward_solution` will project electrodes onto scalp before calculations, wouldn't it be better to do so during `mne.gui.coregistration`, using a button "project" or something like that?

This was we can visualize the real position of electrodes, that might be far from scalp due to digitization issues.

Eric89GXL, 2016-10-26:
@christianbrodbeck any interest? I can make `plot_trans` have such an option first, and thus take care of any necessary refactoring. Then you would just need to cover the `traits`-related tweaks.

Eric89GXL, 2016-10-26:
Actually, what would be cool is `plot_trans(..., eeg_sensors='original' | 'projected' | ['original', 'projected'])`, so show the original locations, projected ones, or both. In the GUI that would be two checkboxes, defaulting to original-only (the current behavior).

christianbrodbeck, 2016-10-27:
I don't really have time (and I'm not working with EEG at the moment) but if it would be very useful in the GUI and all the underlying functions are there I will give it a try

Eric89GXL, 2016-11-18:
@noreun do you think you will have time to do this at some point? You can see how it's done in `plot_trans`

noreun, 2016-11-18:
@Eric89GXL yes, it seems quite simple. I'll give it a shot later tonight.

agramfort, 2017-07-22:
this is done

larsoner, 2018-02-13:
@agramfort this isn't done, you are probably thinking of `plot_alignment` but this issue is for `mne coreg`

PR #4937 (2018-02-23): MRG: Add projection to surface for coreg
This PR:

1. Adds a button to project EEG electrodes to the scalp in `mne coreg`.
2. Fixes a bug where `--low-res-head` didn't always use a low res head (fixed by first looking for `outer_skin.surf` if it exists; `subject-head.fif` is now second, as it can be a dense head).
3. Adds support for normals when surfaces are loaded, enabling hair growing with e.g. `outer_skin.surf` files.
4. Adds support for `~` in the `--subjects-dir` and `--trans` paths using `expanduser`.

@christianbrodbeck do you have time to look and try? Example that shows all of these [... cut]
closed 2018-02-23 by larsoner