mne-tools/mne-python #3702: ENH: plot_topomap, plot_joint & plot: API homogeneity for scalings/scale parameters

kingjr, 2016-10-26:
Small API heterogenity: different plotting functions use similar parameters slightly differently (and named slightly differently)

* scaling: `evoked.plot(scalings=dict())` but `evoked.plot_topomap(scale=dict() | float)`
* units: `evoked.plot(units=dict())` but `evoked.plot_topomap(unit=dict() | str)`
* max amplitude: `evoked.plot(ylim=dict())` but `evoked.plot_topomap(vmin=dict() | float, vmax=dict() | float())`

Changing parameter names can be annoying for deprecation, but I suggest we at least make their input similar (e.g. always accepts both `dict` and `float`).

Additionally, `evoked.plot_joint()` could gather the parameters common to `plot()` and `plot_topomap()` to simplify it. WDTY @jona-sassenhagen ?

jona-sassenhagen, 2016-10-26:
Somewhat related: #3164

jona-sassenhagen, 2016-10-26:
Unity here would be greatly improved IMO.

We must be careful however with how different functions handle different channel types.

jona-sassenhagen, 2016-10-27:
Can we split this into multiple smaller PRs?

jaeilepp, 2016-10-27:
It feels a bit redundant to have ``scaling`` and ``vmin, vmax`` in the first place. Could we get rid of scaling in for instance topomaps or is there a use case where you'd prefer using scaling?

agramfort, 2016-10-27:
to me scaling and vmin and vmax very different usage. I think we should
keep them.

jona-sassenhagen, 2016-10-27:
I think you need scaling if you use for example stats.regression, where you plot t values on data channels.

jona-sassenhagen, 2016-11-29:
Ugh I must get back to this. @jaeilepp can you maybe assign me so I don't forget?

jaeilepp, 2016-11-29:
Done.

jona-sassenhagen, 2016-12-06:
Also: mne.viz.plot_compare_evokeds is missing a bunch of options, like scalings and units.

:(

jona-sassenhagen, 2016-12-06:
@Eric89GXL @agramfort do you have any idea about solving the different names for identical params for different functions (e.g., "unit" vs "units")?

I guess a **kwargs could do, but we never use that (in public functions), so ...

Eric89GXL, 2016-12-07:
Can you give an example of what you are trying to do?

Eric89GXL, 2016-12-07:
Oh it is in the top post.

I think the only clean way to fix it is to deprecate.

PR #4595 (2017-09-25): API: Use scalings
Closes #3702.

This PR:

1. Unifies `scale`->`scalings`
2. Fixes lots of warnings in tests
3. Includes a fix for `plot_topo` that was necessary for tests to pass locally (and for interactivity to work for me at all), so I slightly modified two tutorials so that CircleCI would generate them.

Ready for review/merge from my end.

(One of the last big blockers for 0.15!)
closed 2017-09-25 by agramfort