mne-tools/mne-python #3708: Unused setup_volume_source_space parameters.

jaeilepp, 2016-10-28:
``setup_volume_source_space`` has three params that it is not using ``subject``, ``subjects_dir`` and ``overwrite``.

Eric89GXL, 2016-10-28:
`fname` is also a silly argument, since we have `write_source_spaces`.

I have also never liked that we have `setup_source_space` and `setup_volume_source_space`, the names are not consistent. I wonder if we should take this opportunity to make new functions without the old API cruft. Eventually we could deprecate the old ones, but they could just as easily stick around as wrappers with clear `.. note::`s in the docstring suggesting use of the new functions.

agramfort, 2016-10-28:
what API do you suggest?

Eric89GXL, 2016-10-28:
Thinking out loud after some `git grep` work, generally use `make_` instead of `setup_` in our function names, so we could start there:
```
make_surface_source_space
make_volume_source_space
make_discrete_source_space
```
More explicit, more consistent. And we can shed old, unused, or mis-named arguments.

agramfort, 2016-10-28:
@dengemann any concern?

dengemann, 2016-10-30:
Ok by me, I somehow implicitly felt that "make" is about impure functions that write stuff to disk but it is not consistently the case. Fine to change names if the overall picture is more consistent. Although I hate the resulting deprecation cycle. Let's assume MNE is used in a few years by 10 times as many people. They will be happy.

agramfort, 2016-11-01:
fine with me then.

allow writing in these make_ functions  is also something we should avoid.

we could try however to a default so :

write_source_space(src)

works out of the box.

Eric89GXL, 2016-11-01:
Agreed, the writing should be separate. Within those functions IIRC all they do is call `write_source_spaces` under the hood anyway

agramfort, 2016-11-01:
are you on it?

Eric89GXL, 2016-11-01:
not right now, so feel free if you have time

otherwise, yes I can probably do it this week

Eric89GXL, 2016-11-03:
...actually come to think of it, we probably *should* use `subject` and `subjects_dir`, along with `mri`, to construct the MRI volume to use if possible. In other words, a user should just be able to say `setup_volume_source_space('sample', mri='T1.mgz', subjects_dir=subjects_dir)` and have it work. And we should have it use `overwrite`, that is a bug. So I think instead of this deprecation stuff we should actually have it use the parameters properly / more usefully.

closed 2016-11-22 by jaeilepp