mne-tools/mne-python #3711: raw.plot should start from first_samp

jasmainak, 2016-10-29:
I noticed that `raw.plot()` shows time points from 0 even though `first_samp` is not 0. We should probably change this?

agramfort, 2016-10-29:
indeed

like mne_browse_raw does

Eric89GXL, 2016-10-29:
But raw.times starts from 0 always. So we have to decide where to be
inconsistent.

Is there some backward compatible way we can phase out the first samp
altogether? It seems to cause problems and confusion all the time.

christianbrodbeck, 2016-10-29:
That would be great, but since events are relative to `first_samp` I think any saved events would break...

Eric89GXL, 2016-10-29:
Yeah they would. It would be a pain to deprecate but I think in the long
term we would be happy about it.

One step in that direction would be to add an init argument to RawFIF to
reset or ignore first samp. All other readers already set it to zero, so
that itself would make things consistent. That combined with a long
deprecation cycle to move the default it to True might do it. In a year or
so, by default first samp arguments would be irrelevant and only there for
legacy analyses and files.

agramfort, 2016-10-29:
the benefit of first_samp is that if you crop a file the events are still
valid. I think it's not as stupid as it sound to have a first_samp that
defaults to 0 for vendors that don't support this.

An easy fix is to add a parameter to raw.plot to use or ignore first_samp

deprecation/change of .times attribute is more difficult.

let's be pragmatic here.

jasmainak, 2016-10-29:
I am fine with just fixing the visualization with the `ignore_first_samp` param. I think it would be pedagogical too, in terms of telling the user that the recording starts not at t=0. The only problem is that your `start` parameter can no longer default to `0.0` if there is a `first_samp` ...

Eric89GXL, 2016-10-31:
> the benefit of first_samp is that if you crop a file the events are still valid

Ahh you're right, I thought we weren't doing that. Never mind...

> I am fine with just fixing the visualization with the `ignore_first_samp` param

+1 from me.

Strictly speaking we'd need to have a deprecation cycle to get it to default to showing it relative to the first sample. However, in this case, because `raw.plot` is likely only exclusively for interactive plotting as opposed to making figures (there is no `axes` parameter like our other plotting functions, and it's full of interactive bits) I think we can have it just start plotting relative to `first_samp` by default. It's almost a bug fix anyway, I suppose.

Is there any way out of the inconsistency with `.times` and `raw[:][1]` output? Maybe not worth it?

agramfort, 2016-11-01:
I would say not worth it but I would be fine adding an absolute_times
attribute that would take into account the first_samp

would that be good enough?

Eric89GXL, 2016-11-01:
Yeah, but I'm fine waiting until the next time someone hits a problem with it and hoping it just never comes up again

Eric89GXL, 2016-11-01:
Or @agramfort do you propose that this property should be added instead of changing `raw.plot` with a new kwarg?

agramfort, 2016-11-01:
no I propose to update plot with param to show absolute time.

and eventually in the future add the property absolute_times

Eric89GXL, 2016-11-01:
Sounds good

@jasmainak are you on it?

jasmainak, 2016-11-01:
@jasmainak is in bretagne embracing French lifestyle and spreading the MNE
gospel among tourists. But I'll get to it tomorrow :)

Eric89GXL, 2016-11-14:
@jasmainak you on this? I needed it the other day and didn't have it...

jasmainak, 2016-11-15:
@Eric89GXL it's in my todo list. I can get to it on Thursday if it's not too late for you ...

Eric89GXL, 2016-11-15:
Yeah that's fine

jasmainak, 2016-11-22:
okay, I'm on it now. Sorry for the delay ...

closed 2017-01-13 by jaeilepp