mne-tools/mne-python #3719: Q? HPI in MRI space

kingjr, 2016-11-01:
Hi,

@romquentin and I have a CTF dataset where the head position coils are already registered in MRI space. I think in this case we don't need to do the head shape coregistration.

Is there an easy to use these MRI coordinates directly to build the source space? i.e. to skip `mne.gui.coregistration`?

Thanks!

cc @christianbrodbeck

Eric89GXL, 2016-11-01:
So you have the HPI coil coordinates in only MRI space, or you also have
their digitized locations in head coordinates (with likely slight
mismatches)? I assume you mean the former but want to check.

Do you have any other digitized points? In what coordinate frame?

If all you have is HPI in MRI, and only MEG data, I think you can get away
with using the identity for head to MRI trans, at least for the purposes of
computing the forward model. If you have EEG data with electrodes digitized
in head coordinates, it's a different story.

kingjr, 2016-11-01:
> So you have the HPI coil coordinates in only MRI space, or you also have
their digitized locations in head coordinates (with likely slight
mismatches)? I assume you mean the former but want to check.

 HPI coil coordinates in only MRI space only

> Do you have any other digitized points? In what coordinate frame?

Well, no. They did digitize some headshape, but in a coordinate frame independent from MRI, these head coils, the MEG device... so I don't think this will be any useful.

> If all you have is HPI in MRI, and only MEG data, I think you can get away
with using the identity for head to MRI trans, at least for the purposes of
computing the forward model. 

Ok, so where shall I put the identity? In dev_head_t?

Eric89GXL, 2016-11-01:
> Ok, so where shall I put the identity? In dev_head_t?

That is the MEG device to head transform. Hopefully that one already exists
and is valid, as it is what the coil activations give you during
acquisition.

You want it in effectively head_mri_t, although we usually just call it
`trans`. You can make an identity one with `Transform('head', 'mri')`, and
pass it to functions like the forward code that require it.

kingjr, 2016-11-01:
right, thanks!

closed 2016-11-01 by kingjr

Eric89GXL, 2016-11-01:
Can you track down a place to put this in the docs?

kingjr, 2016-11-01:
That'd be a perfect job for @romquentin :]

romquentin, 2016-12-07:
Hi guys, 
Do I need to use `read_dig_montage` and `epochs.set_montage` before using the identity matrix as the `trans` in my forward model? 
It seems that I cannot run read_dig_montage without hsp argument? (and I do not have any headshape but only the three HPI position already in the MRI space).
Thanks a lot for your help, 
Romain

Eric89GXL, 2016-12-07:
> Do I need to use read_dig_montage and epochs.set_montage before using the identity matrix as the trans in my forward model?

You shouldn't need to for forward modeling, assuming you only have MEG data.

> It seems that I cannot run read_dig_montage without hsp argument? 

It seems like we should support `read_dig_montage` even without the `hsp` argument. This is *probably* a bug...

romquentin, 2016-12-07:
Ok.
But before using `fwd = make_forward_solution(raw.info, fname_trans, src, bem_sol)`, I should add the information somewhere of my HPI coordinates (in head space which is MRI space in my case, that is the reason of using the identity matrix as the trans matrix). I thought read_dig_montage and epochs_set_montage would do that...
Thank you!

The bug when running 
`read_dig_montage(hsp=None, hpi=hpi, elp=elp,
                 point_names=point_names, unit='mm',
                 transform=False,
                 dev_head_t=False)
`
is

TypeError                                 Traceback (most recent call last)
/Applications/anaconda/lib/python2.7/site-packages/IPython/core/formatters.pyc in __call__(self, obj)
    697                 type_pprinters=self.type_printers,
    698                 deferred_pprinters=self.deferred_printers)
--> 699             printer.pretty(obj)
    700             printer.flush()
    701             return stream.getvalue()

/Applications/anaconda/lib/python2.7/site-packages/IPython/lib/pretty.pyc in pretty(self, obj)
    381                             if callable(meth):
    382                                 return meth(obj, self, cycle)
--> 383             return _default_pprint(obj, self, cycle)
    384         finally:
    385             self.end_group()

/Applications/anaconda/lib/python2.7/site-packages/IPython/lib/pretty.pyc in _default_pprint(obj, p, cycle)
    501     if _safe_getattr(klass, '__repr__', None) not in _baseclass_reprs:
    502         # A user-provided repr. Find newlines and replace them with p.break_()
--> 503         _repr_pprint(obj, p, cycle)
    504         return
    505     p.begin_group(1, '<')

/Applications/anaconda/lib/python2.7/site-packages/IPython/lib/pretty.pyc in _repr_pprint(obj, p, cycle)
    692     """A pprint that just redirects to the normal repr function."""
    693     # Find newlines and replace them with p.break_()
--> 694     output = repr(obj)
    695     for idx,output_line in enumerate(output.splitlines()):
    696         if idx:

/Applications/anaconda/lib/python2.7/site-packages/mne/mne-python/mne/channels/montage.pyc in __repr__(self)
    392         """String representation."""
    393         s = '<DigMontage | %d Dig Points, %d HPI points: %s ...>'
--> 394         s %= (len(self.hsp), len(self.point_names),
    395               ', '.join(self.point_names[:3]))
    396         return s

TypeError: object of type 'NoneType' has no len()

Eric89GXL, 2016-12-07:
Are you in latest master? I thought I changed that line recently

romquentin, 2016-12-07:
Just updated and it worked. 
So is `read_dig_montage` and `epochs.set_montage`the good way to add information about HPI coordinates?
Thank you!

Eric89GXL, 2016-12-08:
That should be okay. Can you check it with plot_trans?

Is there any use in having those coordinates? They shouldn't be used for
anything I don't think...

romquentin, 2016-12-09:
Thank you very much. It seems it's working well!
I am using these coordinates because there are the only ones I have... I coregistered before the MEG the head and the MRI with brainsight (which is use for TMS neuronavigation). Although unusual, this navigation system is accurate. The other system that people proposed to me at the MEG center is to take picture of the position of the HPI and I though it was more accurate to have their positions in the MRI coordinate system. 

I do not need to create an identity matrix, this is the default for trans argument in `make_forward_solution` . Let me know if I can still help to put this in the docs?

[image]

agramfort, 2016-12-09:
yes it's better to leave a note in the narrative doc so the discussion from
here does not get lost

romquentin, 2016-12-09:
Ok. Not sure where to leave the note?  Can you send me a link?

agramfort, 2016-12-09:
somewhere in manual eg:

http://martinos.org/mne/dev/manual/index.html#source-localization

or in one of a tutorials

romquentin, 2016-12-13:
Finally, this plot_trans output was a fluke, it did not work correctly with others participants.
I have two questions: 
When I read my epochs, I already have a matrix in `epochs.info['dev_head_t']`. Does it mean that I do not need to compute it in `read_dig_montage`? It is normal that the dev_head_t matrix is not the same originally in the epochs and after added it with dev_head_t=True in read_dig_montage?

Also, when I manually looked at my hpi coordinates in the MRI coordinate system on freeview, my HPIs appear at the good position on the subject MRI. But after `read_dig_montage` and `plot_trans`, HPIs positions (green dot) can be shift relatively to the MRI. From what I understand, it should not be the case if the -trans matrix is the identity one, no?

Thank you very much for your help!

Eric89GXL, 2016-12-13:
> When I read my epochs, I already have a matrix in epochs.info['dev_head_t']

Okay then your head frame is *not* arbitrary, it already exists as something meaningful. This makes sense actually I should have thought about it before. Using `trans=None` will not work :(

> It is normal that the dev_head_t matrix is not the same originally in the epochs and after added it with dev_head_t=True in read_dig_montage?

No, this is weird. It means that the HPI coils that you are passing are not in the correct coordinate frame. One set is supposed to be in head coords and the other in device coords, which gives the `dev_head_t`.

>  when I manually looked at my hpi coordinates in the MRI coordinate system on freeview, my HPIs appear at the good position on the subject MRI. But after read_dig_montage and plot_trans, HPIs positions (green dot) can be shift relatively to the MRI

You should be able to check and see what the coordinates are in the `DigMontage` instance, are they different from what you see in FreeView?

dgwakeman, 2016-12-13:
also, I didn't see any discussion of which "MRI space" you mean. A decent MGZ (generated from dicoms) has at least 3 transformation matrices, all of these could be described as "MRI spaces". The ones I can remember off the top of my head are tk2ras ras2vox. I suspect at least one if not multiple of these will need to be applied to your coordinates.
hth
d

romquentin, 2016-12-13:
@Eric89GXL. Thanks. But I do not have any information in my epoch data about the HPI coils in head coords since the neuronavigation system that I use is not linked to my raw data at any time before trying to apply the montage. So it should be impossible to construct the dev_head_t before applying the montage no? Sorry, How can I check coordinates in DigMontage?
@dgwakeman. The MRI coordinates I am referring to is the RAS coords in freeview (not the TkRegRAS or the surface RAS). 
Thanks a lot!!

romquentin, 2016-12-13:
Ok. When I am running this command:
`dig_montage = read_dig_montage(hsp=None, hpi=hpi, elp=elp,
                               point_names=point_names, unit='mm',
                               transform=False,
                               dev_head_t=True)`
with

hpi=array([[ 0.00407583,  0.10639121, -0.061938  ],
       [-0.06226506,  0.03360433, -0.089045  ],
       [ 0.0627302 ,  0.02532453, -0.088612  ]], dtype=float32)
(corresponding to the values in `epochs.info['hpi_results'][0]['dig_points']`)

elp=array([[  9.912,  87.837,  14.563],
       [-51.649,  20.382, -34.179],
       [ 71.327,  18.049, -38.179]])
(corresponding to my HPI coils position in RAS coords from my neuronavigation system)

point_names = ['NEC', 'LEC', 'REC']

My dig_montage seems to have only hpi informations (but not elp).
`dig_montage`
`<DigMontage | 0 extras (headshape), 3 HPIs, 0 fiducials, 0 channels>`

Hope I am clear enough and thanks a lot again!

dgwakeman, 2016-12-13:
@romquentin , I think there is some more information in some of the CTF files that you aren't detecting. The CTF system must have the location of the HPI coordinates in either device or CTF head coordinate frame (otherwise, it wouldn't be able to measure the head position) in one of the files it reads. 

Essentially all you need to do is calculate the transformation matrix between your HPI coils in RAS space and your HPI coils in either the CTF head or device coordinate frame. Then save that as a trans.fif

romquentin, 2016-12-13:
Yes, epochs.info['dig'] contains information before applying set_montage but I am not sure it is accurate (because at this time I did not give yet any information about HPI position in head or MRI space)
After applying set_montage, epochs.info['dig'] changed and contains now the coils position in RAS coords but with a 10^-3 factor. 
Ok. will try! Thanks!

Eric89GXL, 2016-12-13:
> (because at this time I did not give yet any information about HPI position in head or MRI space)

Are you sure? Somehow the system registered the MEG device<->head transformation, and that usually involves the (digitized, in head coords) HPI coil positions.

romquentin, 2016-12-13:
Yes sure. The brainsight system (neuronavigation system) is not connected at all with the MEG acquisition system so there is no way that my epochs can have information about digitized HPI coil positions before set_montage.

dgwakeman, 2016-12-13:
I get that the brainsight system isn't connected, but there must be some other source of information that the MEG gets information from about the HPI coil position/head coordinate system. Either typed in, or another digitization system, or something.

romquentin, 2016-12-14:
One issue was that I was giving RAS coordinates and not surface RAS in my `elp`. I think mne function use surface RAS during forward modeling..?
I still do not know why I already have epochs.info['dig'] in my data before `set_montage` but I will check tomorrow with the MEG technician. 
I will confirm tomorrow that this strategy works. Thanks a lot again for your help.

romquentin, 2016-12-14:
I confirm that it works (recording my head in different position inside the MEG and reconstruct correctly in plot_trans). This is empirical and I am happy to send you the code if anyone interested to theoretically check or to add a example (even if apparently it is very rare to use neuronavigation system for head positionning). 
Apparently the `epochs.info['dig']` is added during `mne.io.read_raw_ctf` with `_read_pos`. But this function is looking for a .pos file in the ctf ds folder that I do not have. So I still don't know what are the coordinates here. 
Thank you very very much for your help!!

agramfort, 2016-12-15:
>
> Apparently the epochs.info['dig'] is added during mne.io.read_raw_ctf
> with _read_pos. But this function is looking for a .pos file in the ctf
> ds folder that I do not have.
>
this the approach of some CTF sites such the MNI cc @ftadel

ftadel, 2016-12-15:
This is what we do at the MNI: we copy the .pos file into all the .ds folders, it feels natural to have the head points part of each dataset, just like ou would have them in a .fif file.
If you want some examples: use any of the CTF-based sample datasets from the Brainstorm website.

romquentin, 2016-12-15:
Ok. Thanks! Just not sure to understand where values in epochs.info['dig'] come from if there is no .pos file in the .ds folder (@Eric89GXL)?

Eric89GXL, 2016-12-15:
Can you share a sample file that has this behavior? You can email me privately if necessary

romquentin, 2016-12-15:
https://www.dropbox.com/s/e2d5ff3f6eylmpn/MJAZGAQC_wmtask_20161209_01.ds.zip?dl=0

So if I do:
`raw = read_raw_ctf(drop_box_ds_folder)`
`raw.info['dig']` contains
```
[{'coord_frame': 4,
  'ident': 3,
  'kind': 2,
  'r': array([  0.00000000e+00,   1.01319381e-01,  -4.62223187e-33])},
 {'coord_frame': 4,
  'ident': 1,
  'kind': 2,
  'r': array([ -7.40363507e-02,   2.54408175e-19,  -1.07852077e-32])},
 {'coord_frame': 4,
  'ident': 2,
  'kind': 2,
  'r': array([  7.73589972e-02,  -4.50893407e-19,   3.85185989e-32])}]

```
while the .ds folder has now .pos file.

Thanks a lot!

romquentin, 2016-12-15:
Sorry. I mean while the .ds folder does not have .pos file :)

dgwakeman, 2016-12-15:
Did you talk to your MEG tech/site director? There must be some record of the hpi coils in one of the two spaces above or it can't measure the head position.

romquentin, 2016-12-15:
Yes I saw him. What I understood is there is one recording of HPI position in "dewar" space, and they directly calculate the corresponding position in a head CTF space (with the lpa-rpa axis being the Y-axis and the origin at the center of rpa-lpa segment). `read_raw_ctf` takes these coordinates and apply a transformation to bring these coordinates from CTF head to Neuromag head. This information (new coordinates in Neuromag head + transf) is in raw.info['hpi_results']. 
But still no idea of where raw.info['dig'] is coming from. Here at the MEG center, they are using photo to manually position HPIs on the subject MRI. There is no digitized points recording at any time.

dgwakeman, 2016-12-15:
How do they calculate the the lpa and rpa axis if they don't digitize any points? How do they get the head CTF space (not sure what they use)? They must have some digitized point in common with the HPI coils and the fiducials.

What does it mean: "there is one recording of HPI position in "dewar" space"? (how is the HPI position recorded. Do you mean that the put sine waves through and do dipole fits on them?)

romquentin, 2016-12-15:
Sorry maybe I am wrong about what "digitized points" mean. Correct me if I am wrong.
1 - Before and after each MEG block, we localize the head. During that, the HPI coil positions inside the MEG are recording relatives to the "dewar". It gives us HPI coil positions in device space. 
2 - From these coordinates in device space, CTF automatically calculates a "CTF-head" coordinates frame where (it is just a convention I guess) the left-ear coil position and the right-ear coil position forms the y-axis and the nasion is on the x-axis of this second coordinate frame (the "CTF-head").
3 - Until now, there is no "digitized" point, meaning HPI coordinates recordings with a "polhemus-like" system and based on a reference fixed on the subject's head, no?
Again, sorry if I am unclear or if vocabulary is not accurate. I am learning the field :)

Eric89GXL, 2016-12-15:
@romquentin oh you always place the HPI coils at the fiducials, right? Then it could use the localized coords to define both the head coordinate frame, and the head<->device coordinate transformation.

ftadel, 2016-12-15:
> Do you mean that the put sine waves through and do dipole fits on them?
> you always place the HPI coils at the fiducials, right? 

Yes, this is how it works at most CTF sites. In the CTF recordings, the reference points are typically the coils, and the have to mark the coils in the MRI, which is tricky (the Donders has some fancy markers they see in the MRI, but I think most sites don't do that).
At the MNI we digitize both the coils and the anatomical landmarks, save them in the .pos file (copied to the .ds folder), and use them to convert between the CTF coordinates system (based on the coils) and the equivalent based on the anatomical points (and therefor we can mark anatomical points in the MRI, which is much easier)

romquentin, 2016-12-15:
Yes!
It is for that reason that we use the brainsight system to get HPI coil position in "MRI space". 
But I still do not get where the info['dig'] is coming from without .pos file. 
info['dig'] is different from info['hpi_results'].

ftadel, 2016-12-15:
Isn't just the position of the 3 coils coils in the dewar coordinate system (estimated from the coils themselves) ?

romquentin, 2016-12-15:
No :)

dgwakeman, 2016-12-15:
Ahhh... I think they are getting created from the following sequence.
_read_hc
_compose_meas_info
follow that tree and you should find the bits you are looking for

romquentin, 2016-12-15:
Ok. I will! Thanks a lot!

dgwakeman, 2016-12-16:
@romquentin please report back with your findings. This is the type of thing that is good to have noted somewhere (maybe even the docs. But at least maybe code comments).

romquentin, 2016-12-16:
yes. I will investigate where info['dig'] is coming from and find a place to report the necessity to use surface RAS and not RAS coordinate when someone already have HPI positions in "MRI coordinates". Likely on next Monday now.

romquentin, 2016-12-19:
`_pick_isotrack_and_hpi_coils` is creating the `info['hpi_results']` and the `info['dig']`.
The function is taking HPI coordinates in device space and in ctf head space from .hc file. It applies the t_ctf_dev_dev transformation to the coordinates in device space and the result is in `info['hpi_results']` and it applies the `t_ctf_head_head` to the coordinates in ctf head space and the results is in `info['dig']`. If can be confusing to have a info['dig'] without any digitized information but it will be erased if you set a montage with "true" digitized information. 

Maybe you can add this the following sentence in `plot_trans` and `make_forward_model` in the `trans` description: "You can use the identity matrix if you have HPI digitized information already in MRI space. Make sure in this case to use surface RAS coordinates. 

I am happy to do it but not sure how?

Thank you!

Eric89GXL, 2016-12-19:
I'm not actually sure you can do it in this case. For your data, you have a head coordinate frame, and it is defined by these transforms you're talking about. It's probably *not* the same as your MRI coordinate frame.

To get the proper transform for your data, you'd need to fit the "digitized" (from the HPI fitting) points onto the ones that are in your MRI. You should be able to do this using your coreg tool of choice (`coreg` module or `mne_analyze`).

dgwakeman, 2016-12-19:
Well, it should be possible to calculate the MRI-MEG coordinate frame by taking the HPI locations in MRI space and the HPI locations in CTF-HEAD space and calculating the transform ( a cheap way would be to feed the data under the hood to the code that creates this when you select a good fit for the points).

romquentin, 2016-12-19:
Ok...but the result of `plot_trans` seems very accurate. I tested on different blocks moving my head inside the MEG and the reconstruction of my head position seems very coherent. Is it possible that despite of this good reconstruction, something is wrong for the `make_forward_model`?

Eric89GXL, 2016-12-19:
It suggests that your MRI coordinate frame is very similar to the Neuromag
head coordinate frame. Unless you do something in your MRI acquisition or
processing pipeline (which is very possible) to cause this, it might be a
coincidence and it is probably better not to rely on it. You might do
better by establishing the transformation using GUI alignment (or as
pointed out using algebra based on the point locations in MRI space), since
these points are what the device to head transform is based on. You should
find it is close to identity for this subject, but might be more useful for
others.

You have the fiducial coordinates in MRI space right? We have a private
function somewhere to compute the head transformation from fiducial points,
the math is pretty simple.

romquentin, 2016-12-21:
Yes I have the fiducial coordinates in MRI space. You mean compute the head to MRI transform? What is this function? Thank you again!

Eric89GXL, 2016-12-21:
`mne.transforms.get_ras_to_neuromag_trans`