mne-tools/mne-python #3722: mne.io.read_raw_cnt and Neuroscan data problem

kont40, 2016-11-02:
Dear friends,
I have a Neuroscan *.cnt file that contains 18 channel EEG recording with event markers inside, and I am trying to read it with mne.io.read_raw_cnt in the following way:

montage = read_montage('standard_1020','F3, Fz, F4, FC3, FCz, FC4, C3, Cz, C4, CP3, CPz, CP4, P3, Pz, P4, O1, Oz, O2')
raw = mne.io.read_raw_cnt(filepath, montage,'auto',(),'auto', 'auto', 'auto','dd/mm/yy','True','WARNING')

However, I am getting a message: "RuntimeWarning: The following EEG sensors did not have a position specified in the selected montage: ['FZ', 'FCZ', 'CZ', 'CPZ', 'PZ', 'OZ']. Their position has been left untouched."

1) I have no idea how to set up the montage properly - what should I change to have all the electrodes accepted properly?

2) Next, no idea how to get the name of the trigger events channel

3) I am also not sure if I am supplying the rest of the parameters to the mne.io.read_raw_cnt properly

Any advice would be appreciated

jona-sassenhagen, 2016-11-02:
Upper vs. lower case "z"?

kont40, 2016-11-02:
@jona-sassenhagen Thanks, already tried this, didn't help...

agramfort, 2016-11-02:
did you really type:

montage = read_montage('standard_1020','F3, Fz, F4, FC3, FCz, FC4, C3, Cz,
C4, CP3, CPz, CP4, P3, Pz, P4, O1, Oz, O2')

?

second parameter should be a list of str not a str with channels separated
with ,

kont40, 2016-11-02:
@agramfort Indeed, it is my omission! However, it seems the function is clever enough to parse even that wrong input.
Now I tried also each of these:

montage = read_montage('standard_1020',['F3', 'FZ', 'F4', 'FC3', 'FCZ', 'FC4', 'C3', 'CZ', 'C4', 'CP3', 'CPZ', 'CP4', 'P3', 'PZ', 'P4', 'O1', 'OZ', 'O2'])

montage = read_montage('standard_1020',['F3', 'Fz', 'F4', 'FC3', 'FCz', 'FC4', 'C3', 'Cz', 'C4', 'CP3', 'CPz', 'CP4', 'P3', 'PZ', 'P4', 'O1', 'Oz', 'O2'])

montage = read_montage('standard_1020',["F3", "Fz", "F4", "FC3", "FCz", "FC4", "C3", "Cz", "C4", "CP3", "CPz", "CP4", "P3", "PZ", "P4", "O1", "Oz", "O2"])

montage = read_montage('standard_1020',["F3", "FZ", "F4", "FC3", "FCZ", "FC4", "C3", "CZ", "C4", "CP3", "CPZ", "CP4", "P3", "PZ", "P4", "O1", "OZ", "O2"])

and always getting the same result 
"RuntimeWarning: The following EEG sensors did not have a position specified in the selected montage: ['FZ', 'FCZ', 'CZ', 'CPZ', 'PZ', 'OZ']. Their position has been left untouched."

Not sure if this time I supplied the list of strings correctly, though...

agramfort, 2016-11-02:
ok please share a file and a script to replicate the pb

kont40, 2016-11-02:
@agramfort  I have uploaded the files here: http://zhongbaotong.com/mne 
1: database: http://zhongbaotong.com/mne/cnt_database.zip
2. script: http://zhongbaotong.com/mne/erp.py

jaeilepp, 2016-11-03:
@kont40 [this](https://github.com/mne-tools/mne-python/pull/3726) should fix your problem.

jaeilepp, 2016-11-03:
Ignore the previous comment. The problem is that the montage still has the channel names in camelcase. In your case, the fix would be to change the channel names in the montage to uppercase:
```
montage = mne.channels.read_montage('standard_1020')
montage.ch_names = [ch_name.upper() for ch_name in montage.ch_names]
raw = mne.io.read_raw_cnt(filepath, montage,'auto',(),'auto', 'auto', 'auto','dd/mm/yy','True','WARNING')
```

kont40, 2016-11-03:
@jaeilepp Indeed, the last solved the problem and now channels are accepted well. 
I have one last question on that topic:
Now I try to see the channels and use raw.info , is it normal to see only the first 10 channels under 'ch_names ' in the information listing?
===================================
raw.info
Out[2]: 
<Info | 19 non-empty fields
    bads : list | 0 items
    buffer_size_sec : float | 10.0
    ch_names : list | F3, FZ, F4, FC3, FCZ, FC4, C3, CZ, C4, CP3
    chs : list | 21 items (EEG: 18, STIM: 1, EOG: 2)
    comps : list | 0 items
    custom_ref_applied : bool | False
    description : str | 0 items
    dev_head_t : 'mne.transforms.Transform | 3 items
    events : list | 0 items
    filename : str | C:\Users\s.../omisore1.cnt
    highpass : float | 1.0 Hz
    hpi_meas : list | 0 items
    hpi_results : list | 0 items
    lowpass : float | 20.0 Hz
    meas_date : numpy.ndarray | 2016-10-29 23:45:43
    nchan : int | 21
    projs : list | 0 items
    sfreq : float | 1000.0 Hz
    subject_info : dict | 5 items
    acq_pars : NoneType
    acq_stim : NoneType
    ctf_head_t : NoneType
    dev_ctf_t : NoneType
    dig : NoneType
    experimenter : NoneType
    file_id : NoneType
    hpi_subsystem : NoneType
    kit_system_id : NoneType
    line_freq : NoneType
    meas_id : NoneType
    proj_id : NoneType
    proj_name : NoneType
    xplotter_layout : NoneType
>

jaeilepp, 2016-11-03:
Yes, the output is cropped. ``raw.ch_names`` should print all the channels. It should have the ``...`` extension though.

kont40, 2016-11-03:
@jaeilepp ,  @jona-sassenhagen , @agramfort   Thanks! Very happy with the responses and help I received!
Kamen

closed 2016-11-03 by kont40

Eric89GXL, 2016-11-03:
@jaeilepp are you on the `...` change?

jaeilepp, 2016-11-03:
Yep.