mne-tools/mne-python #3723: BUG: Clarify install instructions

Eric89GXL, 2016-11-02:
We should make it clear that people should install Python2 for now, due to Mayavi.

Also, we need to make sure the instructions still work. Apparently Anaconda has been having some problems due to PyQt5 / matplotlib / mayavi junk. I suspect we'll need to add a line, at least for a little while, along the lines of `conda install numpy scipy sklearn mayavi matplotlib` to force dependency resolution. And this stuff should be backported.

Thanks @bloyl for suffering through the existing directions.

mathurinm, 2016-11-02:
I have installed mne-python and been through this recently so I'd be happy to help.
I tried to reproduce my installation steps this morning, but I have an issue when importing numpy (`ImportError: libopenblasp-r0-ea933816.2.18.so: cannot open shared object file: No such file or directory`)

Steps to reproduce:
```
conda create --name mne python=2.7 ipython && source activate mne
conda install numpy scipy mayavi scikit-learn matplotlib
(mne) mathurin@mathurin-HP-EliteBook-840-G3:~$ which ipython
/home/mathurin/anaconda3/envs/mne/bin/ipython
(mne) mathurin@mathurin-HP-EliteBook-840-G3:~$ ipython
Python 2.7.12 |Continuum Analytics, Inc.| (default, Jul  2 2016, 17:42:40) 
Type "copyright", "credits" or "license" for more information.

IPython 5.1.0 -- An enhanced Interactive Python.
?         -> Introduction and overview of IPython's features.
%quickref -> Quick reference.
help      -> Python's own help system.
object?   -> Details about 'object', use 'object??' for extra details.

In [1]: import numpy
---------------------------------------------------------------------------
ImportError                               Traceback (most recent call last)
<ipython-input-1-5a0bd626bb1d> in <module>()
----> 1 import numpy

/home/mathurin/anaconda3/envs/mne/lib/python2.7/site-packages/numpy/__init__.py in <module>()
    178         return loader(*packages, **options)
    179 
--> 180     from . import add_newdocs
    181     __all__ = ['add_newdocs',
    182                'ModuleDeprecationWarning',

/home/mathurin/anaconda3/envs/mne/lib/python2.7/site-packages/numpy/add_newdocs.py in <module>()
     11 from __future__ import division, absolute_import, print_function
     12 
---> 13 from numpy.lib import add_newdoc
     14 
     15 ###############################################################################

/home/mathurin/anaconda3/envs/mne/lib/python2.7/site-packages/numpy/lib/__init__.py in <module>()
      6 from numpy.version import version as __version__
      7 
----> 8 from .type_check import *
      9 from .index_tricks import *
     10 from .function_base import *

/home/mathurin/anaconda3/envs/mne/lib/python2.7/site-packages/numpy/lib/type_check.py in <module>()
      9            'common_type']
     10 
---> 11 import numpy.core.numeric as _nx
     12 from numpy.core.numeric import asarray, asanyarray, array, isnan, \
     13                 obj2sctype, zeros

/home/mathurin/anaconda3/envs/mne/lib/python2.7/site-packages/numpy/core/__init__.py in <module>()
     12 if 'GOTOBLAS_MAIN_FREE' not in os.environ:
     13     os.environ['GOTOBLAS_MAIN_FREE'] = '1'
---> 14 from . import multiarray
     15 os.environ.clear()
     16 os.environ.update(envbak)

ImportError: libopenblasp-r0-ea933816.2.18.so: cannot open shared object file: No such file or directory

```

Contrary to what's mentionned [here](https://github.com/ContinuumIO/anaconda-issues/issues/445), `conda install libgfortran==1`
 does not change the error.
And the versions as shown by pip freeze are:
```
(mne) mathurin@mathurin-HP-EliteBook-840-G3:~$ pip freeze
apptools==4.2.1
backports.shutil-get-terminal-size==1.0.0
configobj==5.0.6
decorator==4.0.10
enum34==1.1.6
envisage==4.4.0
ipython==5.1.0
ipython-genutils==0.1.0
matplotlib==1.4.3
mayavi==4.4.0
mne==0.13
nibabel==2.1.0
nose==1.3.7
numpy==1.9.3
path.py==0.0.0
pathlib2==2.1.0
pexpect==4.0.1
pickleshare==0.7.4
prompt-toolkit==1.0.8
ptyprocess==0.5.1
pyface==4.4.0
Pygments==2.1.3
pyparsing==2.0.3
pysurfer==0.6
python-dateutil==2.5.3
pytz==2016.7
scikit-learn==0.16.1
scipy==0.17.1
simplegeneric==0.8.1
six==1.10.0
traitlets==4.3.1
traits==4.4.0
traitsui==4.4.0
VTK==5.10.1
wcwidth==0.1.7

```

I cannot install numpy=1.10 because of mayavi requirements, nor numpy=1.8 because of fontconfig requirements.

mathurinm, 2016-11-02:
The numpy import can be fixed using numpy=1.9.0 and libgfortran=1
Now `import surfer` fails (surfer is installed with pip, with pip correctly linked to the environment), with error message:
```
/home/mathurin/anaconda3/envs/mne/lib/python2.7/site-packages/traits/trait_base.py in <module>()
     32 from os.path import dirname, exists, join
     33 
---> 34 from . import _py2to3
     35 
     36 # backwards compatibility: trait_base used to provide a patched enumerate

ImportError: cannot import name _py2to3
```

Removing and reinstalling traits as suggested [here](https://github.com/ContinuumIO/anaconda-issues/issues/767) does not solve the problem.

mathurinm, 2016-11-02:
It turns out installing numpy=1.9.0 along with libgfortran==1 allows me to import numpy. I had issues importing surfer, caused by `import traits.api` (`ImportError: cannot import name _py2to3`), which was apparently due to conflicting install of traits=4.5.0 and 4.4.0 in ~/anaconda3/pkgs. As adviced [here](https://github.com/ContinuumIO/anaconda-issues/issues/767) I removed the traits related folders and reinstalled traits=4.4.0 with conda, which solved the issue. But now `import surfer` fails, with error message:
```
/home/mathurin/anaconda3/envs/mne/lib/python2.7/site-packages/tvtk/array_handler.py in <module>()
     26 
     27 # Enthought library imports.
---> 28 from tvtk.array_ext import set_id_type_array
     29 from tvtk.common import is_old_pipeline
     30 

ImportError: numpy.core.multiarray failed to import
```

jasmainak, 2016-11-02:
I have in the past had such problems ... but using pip to upgrade `matplotlib` and `numpy` after installing everything via conda seems to have worked for me

Eric89GXL, 2016-11-02:
I wouldn't recommend using `pip` to install `numpy` in an Anaconda install. Anaconda by default uses MKL, and installing via `pip` will not. I'm a bit surprised this didn't break a lot of stuff.

mathurinm, 2016-11-02:
I can't manage to find out why pysurfer isn't built with the correct version of numpy; 1.9.0 is the only one installed in my environment. I have also tried building pysurfer from source, but I still obtain `RuntimeError: module compiled against API version a but this version of numpy is 9`

mathurinm, 2016-11-03:
I have not found a clean way to install a working version of mayavi using only conda. However I'm posting this as it might help  others in the future: my solution was to install everything with conda, then update mayavi to 4.5 and numpy to 1.11 with pip:

```
conda create --name test_mne python=2.7  matplotlib scipy scikit-learn ipython numpy mayavi
source activate test_mne
pip install -U mayavi numpy
```

It is also worth adding that if I launch `ipython --matplotlib=qt`, the following import crashes
`from mayavi import mlab`
with error `ValueError: API 'QDate' has already been set to version 1`
The workaround is to launch ipython without options, and as first command input 
```
from pyface.qt import QtGui, QtCore
%matplotlib qt
```

mmagnuski, 2016-11-13:
@Eric89GXL 
> We should make it clear that people should install Python2 for now, due to Mayavi.

Are you sure? I installed mayavi on python 3 with conda:
```
conda install -c menpo mayavi
```
and I have no problems with mayavi (except occasional window freezing in jupyter which I also get for matplotlib in [some cases](https://github.com/matplotlib/matplotlib/issues/6675)).

Eric89GXL, 2016-11-14:
I'm not sure if we should recommend using an unofficial repository.
@agramfort WDYT?

mmagnuski, 2016-11-14:
Yeah, it's not official but it was recommended once in mayavi github issue. I can only say that it works for me.

agramfort, 2016-11-14:
I also use an unofficial mayavi build on conda with py35

i tried many things but apparently using vtk7 from satra and mayavi from
pip works on my machine (TM)

christianbrodbeck, 2016-11-17:
This combination seems to work for me (OS X, Python 2.7):

`$ conda install numpy scipy matplotlib mayavi h5py pyqt=4.10 pygments=1`
`$ pip install mne imageio pysurfer`

With the caveat mentioned by @mathurinm  that I have to import `mlab` before `%matplotlib qt`

Eric89GXL, 2017-01-06:
We should probably also break up the Python install page (https://mne-tools.github.io/stable/install_mne_python.html) to have the advanced steps separate. That way the install instructions almost fit on a single page, which is more user-friendly.

christianbrodbeck, 2017-01-11:
Just since this hasn't been mentioned in this thread, I notice that Anaconda released mayavi 4.5, that should make the installation simpler again.

agramfort, 2017-01-11:
do you have a link about his mayavi package?

christianbrodbeck, 2017-01-11:
A link to what @agramfort ?

agramfort, 2017-01-11:
saying that mayavi is now on anaconda in py3k

christianbrodbeck, 2017-01-11:
It's *not* in Python 3, I just noticed it with `conda install mayavi` (and it does not appear if this is run from a py3 env).

agramfort, 2017-01-11:
ok so no update then

christianbrodbeck, 2017-01-11:
the update I’m referring to is 4.4->4.5

closed 2017-01-11 by Eric89GXL

massich, 2018-02-02:
Just for completeness. We recounted the same problem when trying to reproduce [this](https://mne-tools.github.io/dev/auto_tutorials/plot_point_spread.html?highlight=simulation) example.

PR #3892 (2017-01-11): MRG: Update install and contrib docs
Simplifies the install and contrib docs.

Closes #3723.