mne-tools/mne-python #3725: [Question] How to create the Epochs object the right way?

ppasler, 2016-11-02:
Hi guys,

I am trying to create an `Epochs` object from a raw `numpy array` with 14 channel EEG data.
I have to labelled datasets with the same size (`awake` and `drowsy`), I want to use the `Epochs` to create 1 sec 'windows' for further processing - so its not really event-based, but time-based. 
_Am I using the Epochs object right?_

1) I need an event array
I created an array with 1 sec (samplerate is 128) and half awake/drowsy `[[0, 1, 0] .... [n, 1, 1]]`
_Is this correct?_

2) create the Epochs/EpochsArray object
I found two different ways to create it [Creating Epochs Objects](http://martinos.org/mne/dev/auto_tutorials/plot_creating_data_structures.html#creating-epochs-objects)
_Whats the proper way to create an `Epochs object` for my data?_

agramfort, 2016-11-02:
you should use an EpochsArray as explained here:

http://martinos.org/mne/dev/auto_tutorials/plot_creating_data_structures.html#creating-epochs-objects

choldgraf, 2016-11-02:
In either case you need an event array. This is an array of shape `n_events, 3` where the first column is each event's onset (in *indices*) and the last column is an integer unique to each type of event (the middle column I don't really use and just set to 0s, though the docs explain that one more).

With an events array, you can either create an `Epochs` object from a `Raw` object using the `Epochs` class, or if you have a numpy array of shape `n_epochs, n_channels, n_time_per_event` then you can use this, the sampling frequency, and the `events` array to create an events object from scratch.

So it just depends on what form your data is in. It sounds like if you've got continuous recordings that aren't already broken up into event windows, your best bet is to use `mne.io.RawArray` to create a `Raw` object from your array, and then use the events array you create to turn that into an `Epochs` object.

ppasler, 2016-11-03:
thanks @agramfort and @choldgraf,
I have continous data which are not broken up in windows. I tried both ways, but it seems, I get different output. 

```python
array([[a_1, a_2, ... a_n_points], [b_1, b_2, ... b_n_points], ... [n_chan_1, n_chan_2, ... n_chan_n_points]])
```
and should splitted in windows of 128 values (sample rate 128 ~ 1 sec), where each row is an `Epochs`
```python
[
  array([[a_0, a_1, ... a_127], [b_0, b_1, ... b_127], ... [n_chan_0, n_chan_1, ... n_chan_127]]),
  array([[a_128, a_129, ... a_255], [b_128, b_129, ... b_255], ... [n_chan_128, n_chan_129, ... n_chan_255]])
  ...
]
```
This is how I create my events array to split the data by time regulary
```python
e_num     = 1 ... n_events
e_length  = 1 ~ 128 values
e_ class    = 0, 1 ~ awake, drowsy 
event_array = [[0, 1, 0] .... [e_num, 1, 1]]
```

Is my assumption right, that I can use `Epochs` to split my data (simply by time) for processing window by window (e.g. for FFT). Is the creation of the `events` array right?

jaeilepp, 2016-11-03:
Notice that if you have data in shape ``(n_channels, n_samples)`` you should use ``RawArray``. Then you can use that to create the epochs:
```
import mne
info = mne.create_info(...)
raw = mne.io.RawArray(data, info)
epochs = mne.Epochs(raw, events=events, ...)
```
``EpochsArray`` requires data in shape ``(n_epochs, n_channels, n_samples)``.
Sounds like you might also be interested in [fixed length events](http://mne-tools.github.io/dev/generated/mne.make_fixed_length_events.html?highlight=make_fixed_length#mne.make_fixed_length_events).
Hope this helps.

ppasler, 2016-11-03:
Hi @jaeilepp,
thank you, this is what I was looking for! I run it like this:
```python
a_array = mne.make_fixed_length_events(raw, 1)
```
Now I compared the returned `event_array` with my manually created one and noticed some differences:
```python
a_array      | m_array
[0,   0, 1]  | [0, 1, 1]
[128, 0, 1]  | [1, 1, 1]
[256, 0, 1]  | [2, 1, 1]
```
Unclear to me is the following [Creating Epochs objects](http://martinos.org/mne/dev/auto_tutorials/plot_creating_data_structures.html#creating-epochs-objects)

1. The `index of the event` (1. row)
Why isn't it counted 0, 128, 256 instead of 0, 1, 2?
2. The duration (2.row). 
Why is the duration 0? Shouldn't it be 1 - where 1 means one sample ~ 1*sfreq
The documentation is unclear what 'unit' duration has

Now if I have 2 classes I would 
* define start and stop
* call the function two times
* merge the two arrays

This may be stupid questions - thanks for your patience :)

Greetz paul

_What is the proper way to ask such questions? Is GitHub ok?_

choldgraf, 2016-11-03:
Quick answer to your second question: I believe the second column is *not* the length of the event, that's a typo in the docs. It actually contains the value of the stim channel just before this event.

I thought that was fixed a little bit ago though, can somebody confirm?

ppasler, 2016-11-03:
Hey @choldgraf, thanks for the repsonse.

As far as I can see, it is not changed.
https://github.com/mne-tools/mne-python/blob/master/mne/event.py#L782

The creation of the `event_array`:
```Python
    events = np.c_[ts, np.zeros(n_events, dtype=int),
                   id * np.ones(n_events, dtype=int)]
```
The 2. row is always zero

choldgraf, 2016-11-03:
yeah, it's fine for it to be zero (it's what I always use) because it represents a value of a stim channel, not an event length. Check out the docstring for `find_events`: https://github.com/mne-tools/mne-python/blob/master/mne/event.py#L545

We should make it more clear how events are structured with a simple example or something. When I was teaching this to my students it was pretty confusing to them as well.

ppasler, 2016-11-03:
alright, now i got it :)

another thing when I read the doc "The first column contains the event time in samples", This means, as I understand it, it should be counting 0, 1, 2. In my case with sfreq = 128, 1 sample is 128 datapoints, right?

When I am done with this issue I could provide my code and maybe you could make a example of it.

choldgraf, 2016-11-04:
By "event time in samples" I mean "indices of the data". So if you have 4
seconds of data, sampled at 1000Hz, and the event begins at second 2, then
the event time in samples (or indices) would be 2000.

ppasler, 2016-11-04:
@choldgraf This sentence is ambiguous to me - thanks for the clarification!

Another thing cam up to me:
Is there a way create overlapping `Epochs`
```
x x x x|x x x x|x x x x
x x|x x x x|x x x x|x x
```
As I understand it, I would modify my `event_array` 
```python
# no overlapping (with sfreq = 128)
[0,   0, 1]
[128, 0, 1]
[256, 0, 1]
# overlapping
[0,   0, 1]
[64 , 0, 1]
[128, 0, 1]
[196, 0, 1]
[256, 0, 1]
```
Is there a way to apply a window function (hamming or hann)?

agramfort, 2016-11-04:
> Is there a way to apply a window function (hamming or hann)?

use
https://docs.scipy.org/doc/scipy-0.17.0/reference/generated/scipy.signal.hamming.html
and do the multiplications manually applying it to epochs._data

ppasler, 2016-11-04:
@agramfort thanks, so there is no way to do that internally. But this only makes sense if there's a way to make the `Epochs` overlapping, as I would lose information at the edges, right?

agramfort, 2016-11-04:
# overlapping
[0,   0, 1]
[64 , 0, 1]
[128, 0, 1]
[196, 0, 1]
[256, 0, 1]

would give you overlapping Epochs.

the only problem is that you will duplicate the data internally if you preload

ppasler, 2016-11-07:
@agramfort and the others - thanks, works like a charme!
As promised a simple script to document my steps and show my hurdles.

[_tut_creating_fixed_size_epochs.zip](https://github.com/mne-tools/mne-python/files/575171/_tut_creating_fixed_size_epochs.zip)

closed 2016-11-07 by ppasler

agramfort, 2016-11-07:
thanks !

@jaeilepp can you take a few minutes to integrate this to the narrative doc?

thanks heaps