mne-tools/mne-python #3731: [ENH] deep support for annotations

dengemann, 2016-11-03:
What is the current status on support for mne.Annotations?
Are they consistently respected wherever a raw object has them?
We should think about supporting NANs in continuous data, etc., too. This way we would only use epoching when it is necessitated by question.
I fear it is not so easy, my feeling is that NaN are not treated so wholeheartedly throughout python science stack.

dengemann, 2016-11-03:
cc @jaeilepp @jasmainak @Eric89GXL

agramfort, 2016-11-04:
>
> Are they consistently respected wherever a raw object has them?

what do you mean ? have in mind? is it broken for you?

dengemann, 2016-11-04:
I currently don't see from the API if Annotations are used everywhere. For
example when computing a psd welch I would expect that the time windows
marked by Annotations don't have any impact on the estimation. In other
words: never have any impact on the estimation of whatever is computed from
raw. Does my concern get clearer?

agramfort, 2016-11-04:
ok so then I agree than our PSD estimation should use the annotations and
exclude bad segments from the estimation.

anything else?

dengemann, 2016-11-04:
I did not check this. My question is if I read the API terms and conditions
correctly. We then have to where exactly it is not handled. My second point
is that this exclusion approach as we have it in ICA leads to some buffered
/ epochs way of handling internally. I'm not sure if this would always be
desirable. In the case of spectral estimation you don't want to interfer
with the users windowing choices. If someone wants a welch with 50% overlap
and 32 seconds windows for example you don't necessarily want to epoch
internally, etc.

jaeilepp, 2016-11-04:
At the moment, wherever the data is epoched, the annotations are taken into account.

dengemann, 2016-11-04:
> At the moment, wherever the data is epoched, the annotations are taken into account.

Ok but it is not epoched everywhere, this is my point.

dengemann, 2016-11-04:
In other words, if data is epoched or not is an arbitrary implementation choice. But the users should not care about this. If something is annotated it should be ignored.

dengemann, 2016-11-04:
For the particular case of welch it seems we want to implement our own welch that does windowing internally but ignores bad data segments ....

jasmainak, 2016-11-04:
@dengemann are you suggesting that annotations should be treated as NaNs? Annotations need not be "bad" all the time, though ...

dengemann, 2016-11-04:
> @dengemann are you suggesting that annotations should be treated as NaNs? Annotations need not be "bad" all the time, though ..

I think it's less than this that I suggest. If we have 'bad' annotations you want them to be takin into account irrespective of how MNE conceptualises the data. I voice concerns that in this sense, Annotations are not always used. For spectral estimation on raw data it would probably be the best to reimplement welch etc such that a `n_fft` long data segment gets ignored *prior* to windowing and FFT computation. I'm currently exploring this latter case.

agramfort, 2016-11-04:
as you suggest we need to implement our own Welch that ignores windows that
overlap with a bad segment....

what other use case where Annotations should be used but Epochs are not an
option do you foresee?

kaczmarj, 2016-11-14:
> what other use case where Annotations should be used but Epochs are not an
option do you foresee?

Another use case is when one wants to remove noisy portions of data from analysis altogether, similar to the function of [boundary events](https://sccn.ucsd.edu/wiki/Chapter_03:_Event_Processing#Boundary_events) in EEGLAB.

jaeilepp, 2016-11-15:
One option would be just dividing the data into raw instances based on annotations. Then do welch on these and take the weighted average.

Eric89GXL, 2017-02-14:
Deeper support for annotations would take care of #3952, too

Eric89GXL, 2017-02-21:
Just a note that @bmaess (I think) pointed out that for start-stop FIF recordings i.e. with discontinuities (that are now treated as BAD segments by annotations (#4006), we could filter each contiguous segment separately in `raw.filter`.

I'll change the text of the issue to mention use cases that could use better Annotation support.

Eric89GXL, 2017-02-21:
BTW looking at how this could improve our code, I think it should be a major goal for 0.15, or we could even push it to GSoC if we have anyone interested

Eric89GXL, 2017-08-04:
The only open bit from this issue is in (more detail in) #4353

closed 2017-08-04 by Eric89GXL