mne-tools/mne-python #3735: ENH: Plot ECoG data as STC?

Eric89GXL, 2016-11-04:
For ECoG data it's nice to be able to visualize activity on ECoG electrodes on the brain like:

[image]

At least for the brain activity part, this can be accomplished by nearest-neighbor mapping the ECoG electrode onto the cortical surface and then doing some smoothing during `stc.plot`. Then the sensors are an easy addition on top of that.

I'm sure this violates the forward modeling along the lines of OpenMEEG, but @agramfort do you think it's good enough as a first pass for ECoG viz to project it to STC then use our standard functions for plotting?

API proposal is something like `stc = project_ecog_to_stc(evoked)`, but I don't love the name.

Eric89GXL, 2016-11-04:
cc @choldgraf

agramfort, 2016-11-04:
@choldgraf does that provide publishable figures? acceptable by ecog world?

if it's just a hack let's not rush things here.

choldgraf, 2016-11-05:
A few examples from recent ecog papers:

Chang lab:
[image]

Kai Miller et al:
[image]

Schroeder / Zion Golumbic:
[image]

My upcoming paper :D
[image]

Just a few examples. I think it's more common to show scatterplot-like plots as opposed to STC-style plots, which is why I've mostly focused on doing the 3d-to-2d transformation and plotting on top of that. Some folks do the STC-style plot, but perhaps it is more used for things like making grouped-subject statements (e.g. projecting all subjects to MNI space and giving a general idea for where your effects are).

choldgraf, 2016-11-05:
The more I am looking through the literature the more I feel it's less common to make the surface activity-style plots, and more common to make scatterplot-style plots (or more complicated things like doing TFR plots where each TFR is at a grid location, which also require 2D positions of electrodes)

It'd be cool to be able to make a surface activity plot, though it's unclear exactly what is the right way to project electrode activity onto an area of the cortical surface (e.g. my version in `ecogtools` is really dumb and just assumes spherical activation around the electrode).

choldgraf, 2016-11-05:
For scatterplot stuff, and given that @Eric89GXL has just added ecog-style plotting to `plot_trans`, the scatterplot stuff could be accomplished with probably minimal lines of code. It's be something like `plot_trans_screenshot` and would just return an image of the current view in plot_trans, with the spheres removed, and a dictionary of `ch_name: xy` mappings that were pulled from the Montage.

choldgraf, 2016-11-05:
alternatively, eric suggested extending his example to show how to project the 3d points onto a 2d image, and then if enough folks wanted it later on contributing it then.

(sorry for the deluge of responses but if I recall some folks only respond via email which isn't updated if I update the responses here)

Eric89GXL, 2016-11-18:
I think we have some decent stuff with the 2D/3D PR, closing

closed 2016-11-18 by Eric89GXL